espruino/Espruino #842: Some GPIO's are not working and some are not found in ESP8266

ceremcem, 2016-04-14:
On ESP8266, `dump()` function gives the following output: 

```
>dump()
pinMode(D0,"input_pullup");  // GPIO0, works as expected
pinMode(D4,"input_pullup");  // Where is that pin physically?
pinMode(D5,"input_pullup"); // Where is that pin physically?
pinMode(D12,"input_pullup"); // GPIO12, works as expected
pinMode(D13,"input_pullup"); // GPIO13, works as expected
pinMode(D14,"input_pullup"); // GPIO14, works as expected
pinMode(D15,"input_pullup"); // How can we use this pin, since it's always tied to GROUND?
```
This `Pin` object exists and works as expected, but not seen when `dump()`ed: 

```
D2 // GPIO2, works as expected
```

There is no documentation how to use `GPIO18` (`GPIO16` [regarding to another source](http://fabacademy.org/archives/2015/doc/images/esp-03.jpg)) (the `pin#5`). This pin stays at 3.3V constantly.

Additionally, it's not clear that `D4` and `D5` pin objects physically stand for.

MaBecker, 2016-04-15:
Check [ESPRUINO ON ESP8266 WIFI](http://www.espruino.com/EspruinoESP8266) section PINOUT for details.

gfwilliams, 2016-04-15:
Yeah, I think you're getting confused between NodeMCU pinouts (handled with `NodeMCU.XX`) and the board itself?

Please can you ask these questions on the forum though? You're far more likely to get a decent answer there.

tve, 2016-04-16:
D4/gpio4 and D5/gpio5 exist and are available on an esp-12 module, where they are clearly labelled on the back. I use them all the time.

closed 2016-04-16 by tve

ceremcem, 2016-04-16:
Problem is that we use ESP8266-03, which is not found in the page. 
[image]

tve, 2016-04-17:
So?

gfwilliams, 2016-04-17:
Looks to me like you can use all available GPIO... Not sure what the problem is?

ceremcem, 2016-04-17:
Yes, we can use all available GPIO's, 

1. except for `D15`, since we have to short circuit it to `GND` physically.  
2. Moreover,  `dump()` does not show `D2` object (not urgent though)

I understand `D4` and `D5` are available on another version of ESP8266. Existence of `D4` and `D5` is not a matter then. 

While opening this issue the real urgent problem was being unable to use GPIO16 but then we find out that this pin was for "deep sleep" functionality and we have no chance to use it. So you may ignore this issue's urgency at all.