espruino/Espruino #847: Client Socket does not fire onend function in long-duration connections

ceremcem, 2016-04-23:
Hi, 

I'm establishing a TCP connection as a client with my server with the following piece of code: 

```
var clientSocket, connectToServer;
clientSocket = null;
connectToServer = function(){
  console.log("trying to connect to the server...");
  clientSocket = net.connect({
    host: "example.com",
    port: 8080
  }, function(socket){
    console.log("client connected. ");
    socket.on('end', function(){
      console.log("disconnected!");
    });
  });
};
```
If I power the esp8266, it connects to the server. Afterwards I shutdown the server, so "disconnected!" message is printed. 

But if I stay too long (ie. 5 minutes or more), `socket("end"...` callback is not fired and (because) the connection seems still alive:
```
>clientSocket
={ "type": 0, 
  "#onconnect": function (socket) { ... }, 
  "opt": { 
    "host": "example.com", 
    "port": 8080 }, 
  "sckt": 19, "conn": true, 
  "#ondata": function (data) { ... }, 
  "#onend": function () { ... }
 }
>
```

I tried to reset all state table in the firewall, it does not have any effect.

maze1980, 2017-04-16:
Is it possible that there's a timeout on the firewall, dropping the FIN packet? I'd like to see an example with a local server that can reproduce this. And with a callback on "close", too.

MaBecker, 2017-08-10:
check this code, it fires on('close') but does not remove "conn":true

````
HOST = '192.168.194.101';
PORT = 8080;

var net = require('net');
var client;

function dumpSocketInfo(){
   ESP8266.dumpSocketInfo();
}

function logClient(){
  console.log(client);
}

function test(){
  client = net.connect({host: HOST, port: PORT}, function clientConnect(){});

  client.on('close', function onClose(error){
    console.log("-- onClose:",client,error);
  });

  client.on('error', function onError(error){
    console.log('-- onError:',error);
  });

  client.on('connect', function onConnect(event){
    console.log('-- onConnect:',client,event);
  });
}

setTimeout(test,2000);

````

output for a server that is up and then goes down:

````
// server is up
-- onConnect: Socket { "type": 0,
  "#onconnect": [
    function () {},
    function (event) {console.log('-- onConnect:',client,event);}
   ],
  "opt": {
    "host": "192.168.194.101",
    "port": 8080 },
  "sckt": 22,
  "#onclose": function (error) {console.log("-- onClose:",client,error);},
  "#onerror": function (error) {console.log('-- onError:',error);},
  "conn": true } Socket { "type": 0,
  "#onconnect": [
    function () {},
    function (event) {console.log('-- onConnect:',client,event);}
   ],
  "opt": {
    "host": "192.168.194.101",
    "port": 8080 },
  "sckt": 22,
  "#onclose": function (error) {console.log("-- onClose:",client,error);},
  "#onerror": function (error) {console.log('-- onError:',error);},
  "conn": true }

/// server goes down 
-- onClose: Socket { "type": 0,
  "#onconnect": [
    function () {},
    function (event) {console.log('-- onConnect:',client,event);}
   ],
  "opt": {
    "host": "192.168.194.101",
    "port": 8080 },
  "#onclose": function (error) {console.log("-- onClose:",client,error);},
  "#onerror": function (error) {console.log('-- onError:',error);},
  "conn": true } false

// Double check client after 5min
>client
=Socket { "type": 0,
  "#onconnect": [
    function () {},
    function (event) { ... }
   ],
  "opt": {
    "host": "192.168.194.101",
    "port": 8080 },
  "#onclose": function (error) { ... },
  "#onerror": function (error) { ... },
  "conn": true }                 <===== not removed 
````

looks like a bug to me in the ESP8266 code, @gfwilliams  what do you think?

gfwilliams, 2017-08-10:
I don't think `conn` was ever really meant to be set to 0 (if you look in `socketserver.c` there is no code to reset it). It was only used when figuring out when to send the `connected` event.

Looking at the code there's really no way you can get a `close` and not an `end` event either: https://github.com/espruino/Espruino/blob/master/libs/network/socketserver.c#L418

So I guess the original issue is just because there was no `close` event either - as @maze1980 mentioned, maybe there was no `FIN` packet and the default ESP8266 connections don't have a timeout on them? You could test by setting a server up, connecting, and then pulling the ethernet cable out of the server I guess.

However unless we can reproduce this, given it's from April 2016 and there have been no comments since, I'd be tempted to just close it?

closed 2017-08-10 by MaBecker

ceremcem, 2017-08-10:
You are the boss.

maze1980, 2017-08-15:
The original problem was the onClose event that didn't fire, for whatever
reason. And without an example to reproduce this nobody will look at and -
if this is really a bug - can fix it. The issue could be re-opened at any
time.
The "conn": true is not beautiful, but after the onClose event fired you
can't expect I wouldn't expect to have any valid data in your socket
descriptor - cleanup code would be nice, but memory space is very limited.

wilberforce, 2017-08-18:
This issue resolved by @gfwilliams  might have a bearing on this...

https://github.com/espruino/Espruino/issues/1220