espruino/Espruino #851: update ESP IOT SDK

MaBecker, 2016-04-29:
At the moment the Espruino build for ESP is using esp_iot_sdk_v1.5.0.

Tried esp_iot_sdk_v1.5.3 and it works on ESP-12. 
No idea about ESP-01, because I have none.

@tve Any reason to keep  using 1.5.0 ? 

o used sdk files
 - ESP8266_NONOS_SDK_V1.5.3_16_04_18.zip 
 - MDNS_Patch_ESP8266_NONOS_SDK_V1.5.3.zip

o used compiler
 - xtensa-lx106-elf_20151120

o changes

 - Makefile
  - BOOTLOADER = "$(ESP8266_SDK_ROOT)/bin/boot_v1.5.bin"
  - LIBS += -lc -lgcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lmain -lpwm -lcrypto
 - exports
  - ESP8266_SDK_ROOT=/esptools/esp_iot_sdk_v1.5.3
 - flash command section for boot loader
  - 0x0000  boot_v1.5.bin

make result with sdk 1.5.3 (no  CRYPTO )

````
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user2.elf
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  3 .text         00006a64  40100000  40100000  00002e70  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .irom0.text   00066224  40201010  40201010  000098e0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf
  8 -rwxr-xr-x  1 markbecker  staff    2188 28 Apr 23:45 eagle.app.v6.data.bin
824 -rwxr-xr-x  1 markbecker  staff  418340 28 Apr 23:45 eagle.app.v6.irom0text.bin
 24 -rwxr-xr-x  1 markbecker  staff    9472 28 Apr 23:45 eagle.app.v6.rodata.bin
 56 -rwxr-xr-x  1 markbecker  staff   27236 28 Apr 23:45 eagle.app.v6.text.bin
** user1.bin uses  457316 bytes of 491520 available
````

flashing and checking the new build 

`````
>require("ESP8266").getState()
={
  "sdkVersion": "1.5.3(aec24ac9)",
  "cpuFrequency": 160, "freeHeap": 11576, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
>process.memory
=function () { [native code] }
>process.memory();
={ "free": 1570, "usage": 30, "total": 1600, "history": 8 }
`````

tve, 2016-04-29:
The reason I haven't upgraded beyond 1.5.0 yet is that Espressif hasn't made liblwip_536 available for these versions. I've bugged them about it but have not heard back. The normal lwip uses quite a bit more memory so you run out of heap faster.
Is there any specific enhancement you're looking for is .2 or .3?

MaBecker, 2016-04-29:
> liblwip_536

Espressif is including the open source version of liwip since version 1.5.2 (examples/lwip_open_src_template_proj/lwip), can this be used to build the liblwip_536 ? 

> Is there any specific enhancement you're looking for is .2 or .3?

There are a lots of optimization and added API calls  in .1, .2 and .3 
- [1.5.1](http://bbs.espressif.com/viewtopic.php?f=46&t=1609) 
- [1.5.2](http://bbs.espressif.com/viewtopic.php?f=46&t=1702)
- [1.5.3](http://bbs.espressif.com/viewtopic.php?f=46&t=2041)

No, nothing special, but I think with all these optimization it will be a even more efficient and stable version of Espruino on ESP.

wilberforce, 2016-05-09:
Update 4 may:

> We are going to release the new version SDK, ESP8266_NONOS_SDK_V1.5.4 which will have the liblwip_536.

http://bbs.espressif.com/viewtopic.php?t=2104

wilberforce, 2016-06-14:
@tve @MaBecker 

Looks like it's available now....

> Please have a try with the attachment based on ESP8266_NONOS_SDK_V1.5.4.

http://bbs.espressif.com/viewtopic.php?t=2104

MaBecker, 2016-06-14:
yep - compiled last night - not sure if liblwip.a is working with a size of 536

wilberforce, 2016-06-14:
Can you connect to an access point?

Do you need to make any code changes to get it to compile?

MaBecker, 2016-06-23:
@wilberforce  connect to the AP in my network works fine for me
changes: not more than I describe above

tve, 2016-06-23:
looks like this can be closed...

closed 2016-06-23 by tve