espruino/Espruino #852: setTimeout/setInterval  ESP8266.printLog() rouns out of jsvars

HyGy, 2016-04-30:
espruino is running on ESP8266 (1v85.tve_master_cee7141)

This is the example code:

```
setInterval(function () {
    setTimeout(function(){
          console.log("E-Flags: ["+E.getErrorFlags()+"]\n\n");
          console.log(process.memory());
          console.log(ESP8266.getState().freeHeap);
          ESP8266.printLog();
        },
        100);
}, 500);
```

Espruino version:
```
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v85.tve_master_cee7141 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
```

```
>ESP8266.getState()
={
  "sdkVersion": "1.5.0",
  "cpuFrequency": 160, "freeHeap": 10688, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
```

The heap is keep the same.

```
>process.memory()
={ "free": 1136, "usage": 464, "total": 1600, "history": 385 }
```

```
E-Flags: [LOW_MEMORY]
{ "free": 19, "usage": 1581, "total": 1600, "history": 1 }
ERROR: Out of Memory!
{  }
Execution Interrupted during event processing.
at line 3 col 50
          console.log(ESP8266.getState().freeHeap);
                                                 ^
in function called from system
```

MaBecker, 2016-04-30:
produced the same with this code:

````
esp = require("ESP8266");

setInterval(function () {
          console.log("E-Flags: ["+E.getErrorFlags()+"]\n\n");
          console.log(process.memory());
          console.log(esp.getState().freeHeap);
          //esp.printLog();
          esp.readLog();
}, 500);
````

and even without setIntervall:

`````
> console.log(process.memory());
{ "free": 1663, "usage": 37, "total": 1700, "history": 9 }
=undefined
> esp.readLog();
=undefined
> console.log(process.memory());
{ "free": 1661, "usage": 39, "total": 1700, "history": 12 }
=undefined
> esp.readLog();
=undefined
> console.log(process.memory());
{ "free": 1659, "usage": 41, "total": 1700, "history": 12 }
=undefined
`````

wilberforce, 2016-04-30:
```
ESP8266 = require("ESP8266");
setInterval(function () {
          console.log("E-Flags: ["+E.getErrorFlags()+"]\n\n");
          console.log(process.memory());
          console.log(ESP8266.getState().freeHeap);
          //ESP8266.printLog();
}, 500);
```

This works - so it looks like `ESP8266.printLog()` and `ESP8266.readLog()` leak 1 jsVar each call.

wilberforce, 2016-04-30:
https://github.com/espruino/Espruino/blob/db470b6bff59df1c4af17302713a9c873be421a3/targets/esp8266/jswrap_esp8266.c#L160
calls this:
https://github.com/espruino/Espruino/blob/db470b6bff59df1c4af17302713a9c873be421a3/targets/esp8266/jswrap_esp8266.c#L148
 void jswrap_ESP8266_printLog(
) {
  JsVar *line = esp8266_logGetLine();
  while (jsvGetStringLength(line) > 0) {
    jsiConsolePrintStringVar(line);
    line = esp8266_logGetLine();
  }
}
Looks like the *line is not free'd afterwards?

gfwilliams, 2016-05-02:
Yeah - or during the loop.... It should be a pretty easy fix though.

I guess I'd be worried that `esp8266_logGetLine` could be doing something odd as well?

wilberforce, 2016-05-02:
Thanks @gfwilliams 

Yes - during the loop for both functions, and at the end.

https://github.com/espruino/Espruino/blob/2d30d810baba1a00819a69e04df311f9e89b082f/targets/esp8266/log.c#L110
last line:
`  return jsvNewFromString(buf);`

Without knowing the code - what needs to be called to release the var created with ` jsvNewFromString`?

MaBecker, 2016-05-05:
@gfwilliams  I like to understand this, what kind of changes need to be done ?

gfwilliams, 2016-05-05:
Just fixed - take a look at the diff. It's literally just a matter of adding jsvUnLock

closed 2016-05-05 by gfwilliams

wilberforce, 2016-05-05:
Thanks @gfwilliams 

For the readlog case, I think it's still going to leak is it's directly mapped as part of the wrapper:

https://github.com/espruino/Espruino/blob/9d9f2aee7a6ba5c2315b4755bb852bdfd6b4fdf5/targets/esp8266/jswrap_esp8266.c#L155

```
/*JSON{
  "type"     : "staticmethod",
  "class"    : "ESP8266",
  "name"     : "readLog",
  "generate" : "esp8266_logGetLine",
  "returns"  : "String with one line from the log, up to 128 characters long"
}
```
Returns one line from the log or up to 128 characters.

And this won't get released....

gfwilliams, 2016-05-05:
That's fine... I think it's covered somewhere, but:

* If you return a JsVar* from a function, it should have been locked
* If you call a function and get a JsVar*, you should unlock it
* If you get a JsVar* as an argument, and you don't explicitly say in the function/docs you're unlocking it, you shouldn't. eg. `jsvGetInteger` and `jsvGetIntegerAndUnLock`.

wilberforce, 2016-05-05:
Thanks for the clarification.

MaBecker, 2016-05-07:
flashed latest version and tested

`printLog()`  is ok and `readLog()` is still leaking.

test code for `readLog()`

````
esp = require("ESP8266");

setInterval(function () {
          console.log(process.memory());
          esp.readLog();
}, 1000);
````

console output:

````
{ "free": 1649, "usage": 51, "total": 1700, "history": 17 }
{ "free": 1647, "usage": 53, "total": 1700, "history": 17 }
{ "free": 1645, "usage": 55, "total": 1700, "history": 17 }
{ "free": 1641, "usage": 59, "total": 1700, "history": 17 }
......
````

test code for `printLog()`

````
esp = require("ESP8266");

setInterval(function () {
          console.log(process.memory());
          esp.printLog();
}, 1000);
````

console output:

````
{ "free": 1649, "usage": 51, "total": 1700, "history": 17 }
{ "free": 1649, "usage": 51, "total": 1700, "history": 17 }
{ "free": 1649, "usage": 51, "total": 1700, "history": 17 }
{ "free": 1649, "usage": 51, "total": 1700, "history": 17 }
....
````

gfwilliams, 2016-05-09:
Are you sure the build is up to date? If you're using @tve's branch he may not have updated to the most recent changes from master.

MaBecker, 2016-05-09:
YES - latest master version - see GIT_COMMIT

````
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v85.MaBe_master_b7efd3e Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016

>process.env.GIT_COMMIT
="b7efd3e9bfa2bf7f16180e3c1ef53c94ab6ce908"

````

commit 9d9f2aee (2016-05-05): Fix ESP8266 printLog memory leak (fix #852)