espruino/Espruino #853: analogRead pin connected to GND on ESP8266 no 0

HyGy, 2016-05-02:
Hi,

I connected A0 to GND.

```
```

gfwilliams, 2016-05-04:
Does it produce more or less the right output when connected to a potentiometer? It might just be that it's not very accurate... Espruino just calls into ESP8266 code, so I don't see there's anything we could do to fix it.

MaBecker, 2016-05-04:
This is what the ESP on my side shows when connected adc pin to ground and calling some reads: 

````
setInterval(function(){
console.log(analogRead());
}, 2000);

/* output 
0.00195503421
0.00293255131
.....
0.00293255131
*/
````

and here is a nice article to analog read on ESPs: 

http://www.esp8266-projects.com/2015/03/internal-adc-esp8266.html

wilberforce, 2016-05-05:
https://github.com/espruino/Espruino/blob/master/targets/esp8266/jshardware.c#L457

Looking at this it looks like analogue read needs a pin to be supplied....

HyGy, 2016-05-05:
@wilberforce 

I'm using this way on the real place:

```
var mq = require("MQ135").connect(NodeMCU.A0);
```

but as I know the analogRead on esp8266 defaults to NodeMCU.A0. And when I try to read manully with analogRead, and breathing on the sensor, i get higher values, and then it falls back. So it need to be working without any param on ESP.

@tve thinks there is stg with the wifi interrupts:
 
> Thorsten von Eicken @tve 09:31
> IIRC there is an issue with wifi interrupts and ADC. I read that a long time ago and forget the details...

wilberforce, 2016-06-20:
I think the 1.5.4 sdk might sort this out

https://github.com/espruino/Espruino/issues/851#issuecomment-225858297

MaBecker, 2016-06-27:
1.5.4 offers an new function for fast analog read, but you have to shut down wifi :-( 

Test results with SDK 1.5.4 and function analogRead(); on ESP8266-12E

analog input floating: 1.00097751710

grounded: 0.00195503421 :  2mV - 0.00391006842 : 4mV

Can*t see any jumps like HyGuy reported.

HyGy, 2016-06-27:
I'll test it when I have time.

wilberforce, 2016-07-03:
@HyGy 
I've tried it, sorry to say it has not made a difference :-(

MaBecker, 2016-07-03:
@HyGy  & @wilberforce 

what type of ESP are you using ?
 
How can I reproduce this jumps ?

HyGy, 2016-07-04:
nodemcu :) don't know I just connected to the gnd nothing else

MaBecker, 2016-07-04:
ok - which ESP is mounted?

this is the model I use AI-Thinker ESP8266 ESP-12E and 12F

HyGy, 2016-07-09:
I flashed the latest version espruino_1v86.3_esp8266 and this is the same.

Maybe what is a new information, that when I disconnect from wifi, it can read stable values.

MaBecker, 2017-07-21:
Lets sum this up

ESP8266 is not reliable to measure analog input, so use additional i2c AD components

and close it.

closed 2017-07-21 by MaBecker