espruino/Espruino #855: i2c returns values for a non existing devices

MaBecker, 2016-05-07:
testing with some i2c devices and using a scanner function publish on the forum [i2cdetect](http://forum.espruino.com/conversations/278556/)

I2C.readFrom(address, quantity) return a value even without a device on the i2c bus 

````
I2C1.setup({sda: D4, scl: D5} );

// 0x52 is not existing
>I2C1.readFrom(0x52,1);
ERROR: No ACK
=new Uint8Array([200])
> 

// 0x3c is existing
>I2C1.readFrom(0x3c,1);
=new Uint8Array([65])
````

Espruino/targets/esp8266/jshardware.c

https://github.com/espruino/Espruino/blob/master/targets/esp8266/jshardware.c#L925

gfwilliams, 2016-05-09:
Is this a bug? As far as I can tell this happens on all devices. I guess maybe it should just throw an exception?

MaBecker, 2016-05-09:
Yes - exception, so we can use try and catch

MaBecker, 2016-05-09:
No bug, works as coded as far I can see.

Not idea about side effects of jsExceptionHere() in this case, will try now.

@tve  what do you think about that ?

MaBecker, 2016-05-10:
changes in  jshI2CWrite()

````
...
error:
  i2c_master_stop();
  //jsError("No ACK");
  jsExceptionHere(JSET_INTERNALERROR, "I2CWrite : No ACK %d\n", ack);
}
````
and in  jshI2CRead()
````
...
error:
  i2c_master_stop();
  //jsError("No ACK");
  jsExceptionHere(JSET_INTERNALERROR, "I2CRead : No ACK %d\n", ack);
}
````

Now try{ } catch{}  is working for I2C.readFrom() and I2C.writeTo()

a i2cdetect javascript function could look like this: 

````
function isDeviceOnBus(i2c, addr) {
  try {  
    return i2c.readFrom(addr,1);
  } 
  catch(err) { 
    return -1;
  }
}

function i2cdetect( i2c, first, last ) {
  if (typeof first === "undefined") first = 0x03;
  if (typeof (last) === "undefined") last = 0x77;
  
  print( "     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f" );
  for (var upper = 0; upper < 8; ++upper) {
    var line = upper + "0: ";
    for (var lower = 0; lower < 16; ++lower) {
      var address = (upper << 4) + lower;
      // Skip unwanted addresses
      if ((address < first) || (address > last)) {
        line += "   ";
        continue;
      }
      //console.log(isDeviceOnBus(i2c,address));
      // if (isDeviceOnBus(i2c,address) instanceof Array ) {
      // if (Array.isArray(isDeviceOnBus(i2c,address))) {
      if ( ! isDeviceOnBus(i2c,address) ) 
        line += (address + 0x100).toString( 16 ).substr( -2 )+" ";
      else 
        line += "-- ";
    }
    print( line );
  }
}

setTimeout(function(){i2cdetect( I2C1, 0x03, 0x77);},1000);

/* output - 

attached one  MLX90614

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- 5a -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

*/
````

not happy about this `if ( ! isDeviceOnBus(i2c,address) ) `

@gfwilliams  can you please suggest  a elegant way, because this is confusing  ?

gfwilliams, 2016-05-10:
Thanks for the suggested changes...

I'm not sure what you expect instead of `!isDeviceOnBus(i2c,address)` - as far as I know that's basically what you'd have to do to see if a device is there.

MaBecker, 2016-05-10:
sorry for not describing it clear.

Found a better readable solution: `if ( isDeviceOnBus(i2c,address) !== -1 )` 

- -1: no device at address

and     
    
- Uint8Array() : found a device at address

gfwilliams, 2016-05-10:
Ahh - why not just:

```
function isDeviceOnBus(i2c, addr) {
  try {  
    i2c.readFrom(addr,1);
    return true;
  } catch(err) { 
    return false;
  }
}
```

MaBecker, 2016-05-10:
That's it - thanks ! 

so now I use this  simple if :  

`if (isDeviceOnBus(i2c,address))`

tve, 2016-06-23:
Is there something to fix here or can this ticket be closed?

MaBecker, 2016-06-23:
@tve: something to fix

change: jsError(...) to jsExceptionHere(..) in  jshI2CRead() and  jshI2CWrite()
reason: to be able implement something like i2cdetect - see .js code above 

Or is there a better way to accomplish this ?

gfwilliams, 2016-06-23:
Sounds good to me.

MaBecker, 2016-06-23:
created pull request #868

closed 2016-06-23 by MaBecker