espruino/Espruino #857: Wifi.stopAP() is not stopping the AP

MaBecker, 2016-05-11:
try to stop AP using Wifi.stopAP() but it is still up.

````
var Wifi = require("Wifi");
Wifi.stopAP(function(){
  console.log("stopped AP");
 console.log(wifi.getAPDetails());
});

/* output 

stopped AP
{
  "status": "enabled",
  "authMode": "open",
  "hidden": false, "maxConn": 4,
  "ssid": "ESP_05C100",
  "password": "",
  "savedSsid": null,
  "stations": [  ]
 }

*/
````

running on ESP with latest build version from Espruino master 
GIT_COMMIT": "b7efd3e9bfa2bf7f16180e3c1ef53c94ab6ce908

MaBecker, 2016-05-11:
Further info:

You can only stop the ESP AP if connected as Station to another AP.

Is this how ESP works or is this how  the wifi is implemented for Espruino ?

MaBecker, 2016-06-23:
@tve any comment ?

tve, 2016-06-23:
nope, sounds fishy to me, but I didn't test it 'cause I didn't have a laptop handy to join the AP side and cut STA...

tve, 2016-07-24:
Ah, I didn't search well enough and thought it was a forum thread and not a ticket. Grr.

tve, 2016-07-24:
After the initial flashing when Espruino first comes up wifi.stopAP() has no effect (the AP stays on). Once the STA has been started or other wifi settings have been changed the AP can be turned off. Looking at the code this seems to be a bug in the Espressif SDK (1.5.X) and is really more of a nuisance than anything else given that after the initial problem the function works fine.

MaBecker, 2016-07-26:
this will stop station:

require("Wifi").connect("", {password:""});

now it is possible to stop the ap

tve, 2016-07-26:
The proper way to stop the STA is `require("Wifi").disconnect()`

tve, 2016-09-17:
This issue is a bug in the Espressif SDK. I recommend closing this and expecting that perhaps some day Espressif will fix it in the SDK.

MaBecker, 2016-09-23:
tested again with sdk 2.0 - works 

````
>Wifi.getAPDetails();
={
  "status": "enabled",
  "authMode": "open",
  "hidden": false, "maxConn": 4,
  "ssid": "shyFi",
  "password": "",
  "savedSsid": null,
  "stations": [  ]
 }
> Wifi.stopAP();
=undefined
>Wifi.getAPDetails();
={
  "status": "disabled",
  "authMode": "open",
  "hidden": false, "maxConn": 4,
  "ssid": "shyFi",
  "password": "",
  "savedSsid": null }
> 
````

MaBecker, 2016-10-05:
@tve can you confirm ? if yes - than this can be closed

tve, 2016-10-09:
Yup, just tried it an it's fixed!

closed 2016-10-09 by tve