espruino/Espruino #858: Function "toUTCString" not found!

The-Alchemist, 2016-05-12:
Based on http://www.espruino.com/Reference#Date, it seems ``Date.toUTCString()`` should exist.

Maybe I'm doing it wrong?

```
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v85 Copyright 2016 G.Williams
>echo(0);
=undefined
>Date.toUTCString()
Uncaught Error: Function "toUTCString" not found!
 at line 1 col 6
Date.toUTCString()
     ^
> 
```

gfwilliams, 2016-05-12:
Hi - yeah, if you try this in your web browser you get the same error.

What you need is actually an instantiated date object (not just the `Date` class). eg: `(new Date()).toUTCString()`

closed 2016-05-12 by gfwilliams

The-Alchemist, 2016-05-12:
Ah, thanks for the heads up, I must have mis-read the docs.

gfwilliams, 2016-05-12:
It can be a bit confusing as the Espruino docs are a bit sparse and often lack examples - if there's link to `View MDN documentation` then it's best to click that - Mozilla document functionality really well.