espruino/Espruino #859: Ability to reset watchdog timer

gfwilliams, 2016-05-17:
http://forum.espruino.com/conversations/782/#comment12989310

`E.enableWatchdog(timeout_in_seconds)` should maybe be:

`E.enableWatchdog(timeout_in_seconds, auto(bool=true) )`

Then `E.kickWatchdog()` could be added, so you could set `auto` to false (in which case the interpreter didn't reset the watchdog). You could then use  `E.kickWatchdog()` from JS to manually kick it.

closed 2016-07-05 by gfwilliams

gfwilliams, 2016-07-05:
No problem - as discussed before this doesn't work in ESP8266 though, because it only has one watchdog timer, which is used by the internal WiFi stuff.

ceremcem, 2016-07-05:
Well, this hurt :D We found some cute watchdog ic's for this kind of small tasks for ESP8266 and decided that a central supervisor unit (something more than a supervisor ic, such as a small MCU) would better suite in bigger (more critical) systems because we may need more extensive measures, such as re-flashing the ESP (and/or the other(s)). Anyway, thank you for your support :)

commit 35e33374 (2016-07-05): Added  to allow you to keep your JavaScript running - not just the interpreter (fix #859)