espruino/Espruino #860: `dump()` causes reset on ESP8266

ceremcem, 2016-05-22:
Hi, 

I can use every function of ESP8266-12 as expected, except for a `dump()` call causes reset: 

```
...
function unpack(t) {var e,n;try{if(e=JSON.parse(t.trim()),void 0===e)throw null}
function merge(t) {var e,n,o,i,r,f;for(e=slice$.call(arguments,1),n=0,o=e.lengt}
var ser = ASSERT(jsvGetLocks(var) < 15) FAILED AT src/jsvar.c:556
  #1[r2,l2] Object { 
    #2[r1,l2] ASSERT(jsvGetLocks(var) < 15) FAILED AT src/jsvar.c:556
REBOOTING!
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0
2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
...
```

gfwilliams, 2016-05-23:
Looks like this is dependent on which code is uploaded? Could you provide the code that you uploaded and saved first?

ceremcem, 2016-05-23:
[app.zip](https://github.com/espruino/Espruino/files/277693/app.zip)

tve, 2016-06-22:
The basic issue here is either deadlock or watchdog reset. Anything that outputs more than a few KB on the console at once will deadlock because the network code never gets a chance to run (I forget the details). If you only output to serial you end up WDT resetting 'cause the Wifi gets locked out in the busy loop printing.

ceremcem, 2016-06-22:
Ok, then let me see if I can implement a `dump()` function which will take a break after a few KB and continue later...

gfwilliams, 2016-06-22:
> which will take a break after a few KB and continue later...

That won't be easy to do - it's called from the interpreter, so the interpreter would have to be able to start again in the same place.

If you have enough RAM you could try `E.dumpStr()` which dumps everything to a string - so you can then choose what you want to do with it - for instance outputting it only every so often.

ceremcem, 2016-06-22:
I thought I could read the rom manually, part by part, then print it to the console... But this really doesn't hurt as you know the reason and as it's not a bug.

closed 2016-06-22 by ceremcem