espruino/Espruino #861: CPU load when client socket open

manuel-serrano, 2016-05-27:
Dear all,

I'm currently testing Espruino on a Linux box. On that configuration, from the moment a client socket is open and starts receiving data, the cpu load goes up to 100%. I quick look at the C code has not allowed me to understand what's wrong in the Espruino event loop. Your help would be appreciated.

Thanks in advance.

gfwilliams, 2016-05-27:
Ahh, it'll be that `socketIdle`, and hence `jswrap_net_idle` return true - meaning they have been busy. So the interpreter doesn't sleep but instead loops around again.

Unfortunately this polling is generally needed in an embedded system. In Linux I guess you'd do some kind of blocking select call, but that's difficult as you want to wake up not just on network activity, but on other user inputs too.

manuel-serrano, 2016-05-27:
Dear Gordon

First of all, let me thank you for your response.

> Ahh, it'll be that `socketIdle`, and hence `jswrap_net_idle` return true - meaning they have been busy. So the interpreter doesn't sleep but instead loops around again.
>
> Unfortunately this polling [...]
I'm not sure to fully understand all the implications of your answer.
Could you confirm that the following understanding is correct?  Is it
correct to think that only Linux suffers from that problem? In other
words, is it true that this problem I see on my Linux box won't affect
an embedded system?

Thanks in advance,

-- 
Manuel

gfwilliams, 2016-05-27:
You'll get this happening on any Linux-based system with the code as it is.

It's worth noting that if there is JavaScript that needs to run, it's still run at the same speed as if there was no network connection - the 100% CPU is merely Espruino going around the idle loop.

On properly embedded things like microcontrollers, it means that Espruino won't put the micro to sleep when there is an active connection. However generally a wifi adaptor might take 100mA or more with an active connection, so the ~20mA Espruino will be drawing isn't a big deal.

What kind of device were you thinking of running on? It would be pretty trivial to change the code to not do this on Linux systems.

manuel-serrano, 2016-05-27:
> You'll get this happening on any Linux-based system with the code as it is.
>
> It's worth noting that if there is JavaScript that needs to run, it's still run at the same speed as if there was no [...]
We are considering different devices. In particular something based on a
STM32L4 and something equipped with a STM32F7.

-- 
Manuel

gfwilliams, 2016-05-27:
Well, on those kind of systems, Espruino is basically taking 100% CPU all the time (because it is the only thing running). It's just a matter of whether it sleeps or not.

Just a quick note: if you're planning on using Espruino on your own boards, you get to use Espruino for free because it's Open Source - but that doesn't mean you get free tech support :)