espruino/Espruino #870: ASSERT(jsvGetLocks(var) < 15) FAILED in a websocket client application

xiangjunhuang, 2016-07-07:
Hi,
Assertion crash happens when I running a websocket client app. Same code is able to work OK on a 1v85 binary release.

Board: ESP8266-12
Firmware version: $ git rev-parse HEAD
c13891f0218ceab9901ab01e357671c151958d12

I'm running a web socket client application using below code base, and using a PC python script to send periodic message to ESP8266 every second:

var WebSocket = require("ws");

print(process.memory());

var host = "192.168.1.5";
var ws = new WebSocket(host,{
  port: 9090,
  protocolVersion: 13,
  origin: 'Espruino',
  keepAlive: 60
});

ws.on('open', function() {
  console.log("Connected to server.");
});

var msg_cnt=0;
ws.on('message', function(msg) {
  msg_cnt+=1;
  print("MSG: " + msg_cnt);
});

ws.on('close', function(){
  console.log("socket closed.");
});

Output log on webIDE:
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v85.226 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>echo(0);
{ "free": 1449, "usage": 251, "total": 1700, "history": 229 }
MSG: 1
{ "free": 1300, "usage": 400, "total": 1700, "history": 284 }
=undefined
MSG: 2
MSG: 3
MSG: 4
MSG: 5
MSG: 6
MSG: 7
MSG: 8
MSG: 9
MSG: 10
MSG: 11
ASSERT(jsvGetLocks(var) < 15) FAILED AT src/jsvar.c:570
  #1[r1,l2] Object {
    #2[r1,l2] ASSERT(jsvGetLocks(var) < 15) FAILED AT src/jsvar.c:556
REBOOTING!
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 1856, room 16
tail 0
chksum 0x63
load 0x3ffe8000, len 776, room 8
tail 0
chksum 0x02
load 0x3ffe8310, len 552, room 8
tail 0
chksum 0x79
csum 0x79
2nd boot version : 1.5
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
ã{dÄlìd`Äã;$ìdl`ãslÄll`sLoading 1622 bytes from flash...
Running onInit()...
I'm in onInit()...

closed 2016-07-25 by gfwilliams

useapp, 2021-02-21:
the same error in esp32 espruino 2v08

MaBecker, 2021-02-22:
> the same error in esp32 espruino 2v08

Please share some details.

commit 4fba95ec (2016-07-25): Fix 'lock overflow' when calling methods with 'this' bound (fix #870, fix #885)