espruino/Espruino #876: Stack and Heap sizes for nRF51/nRF52

mjdietzx, 2016-07-11:
They should be defined as:

__STACK_SIZE
__HEAP_SIZE

(But the assembler needs to get these pre-processor definitions).

In boards.py probably. Currently we are modifying the gcc_startup_nrf51/2.s files and this will need to be redone every time the SDK is updated.

gfwilliams, 2016-07-11:
Wow - really? That sounds like something that should really be fixed on the nRF SDK side - I haven't heard of the initialisation code needing that before - can't it usually just be got from the linker?

I'm not sure where it'd go, but my vote is somewhere more global. I mean, if we have to update every single nRF51 and nRF52-based board's py file, surely that's worse than just changing the `.s`?

mjdietzx, 2016-07-11:
That's a good point, I think it should be fixed in nRF SDK. Will make a task and see if this can be gotten from linker.

closed 2016-07-11 by mjdietzx