espruino/Espruino #877: Figure out the required stack size for nRF51/nRF52

mjdietzx, 2016-07-11:
and reserve this in the Linker scripts.

and then we can figure out the optimal number of variables in BOARDS.py

gfwilliams, 2016-07-11:
On other platforms we just use all remaining memory as stack (there are actually runtime checks to ensure we don't allow the user to recurse so far they use it all up). Having said this it'd be nice to specify a (1kB?) minimum size?

The current method of determining free stack/heap by tweaking variables is a bit janky, but it's useful to do per-board. For instance on ESP8266 some people do different builds depending on whether they think they'll need more heap space for open sockets, or more variables for code.

Also there are some trade-offs with variable numbers... The second you get over 1023 variables the memory needed goes up a load as the addresses can no longer be packed into bytes.

I think at the end of the day it's something that folks will have to set up on a per-board basis to give the best balance. We could come up with some script that looked at the `.lst` file and suggested variable numbers as a good starting point though.

wilberforce, 2016-07-11:
Would it be possible to save the number of vars in flash, and init this chunk of memory on startup?

I realise this might be over simplifying the situation!

gfwilliams, 2016-07-11:
Not sure I understand - you want to have the number of variables configurable at runtime?

It's possible I think, but within some limits - for instance the code has to be compiled for 12 or 16 byte variables, trying to do that on demand would be virtually impossible.

Thing is, who is really going to know enough about what's happening to actually do this? And if they do, they're probably clued up enough to just compile their own binary.

closed 2017-06-08 by gfwilliams