espruino/Espruino #878: Reserve FLASH space for Espruino to save to nrf51/nrf52

mjdietzx, 2016-07-11:
this flash should be reserved in Linker and 'saved_code' & 'flash_available' changed in the BOARDS files.

gfwilliams, 2016-07-11:
I think to do this properly, we really want to find a way of doing it in the linker for most other platforms as well? If it's just for nRF it's probably more confusing than the current situation of specifying it in board.py

mjdietzx, 2016-07-11:
OK, so basically right now we don't specifically reserve this FLASH in the linker but just set it aside in BOARDS.py. That sounds OK to me, i didn't know that is how it was done for all platforms, thought nRF was the outlier

gfwilliams, 2016-07-11:
Yeah... it's a bit hacky, but it works. I'd would like to get it reserved in the linker at some point though.

If you've got any ideas I'm all ears - maybe we could have some const variable we define, with some attributes that stick it at the correct location in flash memory? If we can do it without touching the linker file then it means we don't have to tweak every platform to get it to work.

gfwilliams, 2016-09-12:
Closing this - all seems to work fine now.

closed 2016-09-12 by gfwilliams