espruino/Espruino #885: jsvLock > 15 assertion due to socket recv callback

tve, 2016-07-25:
The attached program hits the assertion on my linux build (must be built without RELEASE=1). See http://forum.espruino.com/comments/13118945/ and the message after it for details. Running the program requires an MQTT server running at the hostname specified in line 90. The crash output looks like this for me (note that I tweaked the ASSERT):

```
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
MQTT: test/44 Hello world
>#291[r5,l15] Var trace: ÿ.HttpCC["0"ASSERT(jsvGetLocks(var)>0) FAILED AT src/jsvar.c:616
  #1[r1,l2] Object {
    #2[r1,l2] Name String [1 blocks] "\xFF"      #3[r1,l3] Object {
        #6[r1,l2] Name String [1 blocks] "timers"          #7[r2,l2] Array(2) [
            #339[r1,l2] Name Integer 1              #320[r1,l2] Object {
                #278[r1,l2] Name String [1 blocks] "time"= int 999741
                #311[r1,l2] Name String [1 blocks] "interval"= int 1000000
                #308[r1,l2] Name String [1 blocks] "callback"                  #196[r2,l2] ...
 
              }
          ]
        #8[r1,l2] Name String [1 blocks] "watches"          #9[r2,l2] Array(0) [ ]
        #11[r1,l2] Name String [1 blocks] "net"          #10[r1,l3] String [1 blocks] "\x00\x00\x00\x00\x00\xFF\xFF\xFF"
        #13[r1,l2] Name String [1 blocks] "history"          #14[r1,l2] Array(7) [
            #15[r1,l2] Name Integer 0              #12[r1,l2] String [1 blocks] "echo(0);"
...
            #317[r1,l2] Name Integer 6              #313[r1,l2] String [1 blocks] "echo(1);"
          ]
        #275[r1,l2] Name String [1 blocks] "modules"          #276[r1,l2] Object {
            #274[r1,l2] Name String [1 blocks] "net"              #277[r1,l2] NativeFunction 0x4270e5 (0) { }
          }
        #282[r1,l2] Name String [1 blocks] "HttpCC"          #283[r1,l3] Array(1) [
            #297[r1,l2] Name Integer 0              #291[r5,l1] ...
 
          ]
      }
    #30[r1,l2] Name String [1 blocks] "MQTT"      #29[r2,l1] Function {
```

[pingpong.zip](https://github.com/espruino/Espruino/files/380794/pingpong.zip)

tve, 2016-07-25:
Is there an official linux build that I can verify this issue with?

gfwilliams, 2016-07-25:
There isn't - I think there's a Pi version? I'd be pretty certain it's reproducible though - I'll give it a go.

gfwilliams, 2016-07-25:
Yes - just reproduced...

gfwilliams, 2016-07-25:
Ok, think I've got it. It was to do with binding `this` in a function and then calling it on something that also had `this` bound.

commit 4fba95ec (2016-07-25): Fix 'lock overflow' when calling methods with 'this' bound (fix #870, fix #885)
closed 2016-07-25 by gfwilliams