espruino/Espruino #891: E.setBootCode() causes constant reboots on ESP8266

ceremcem, 2016-07-31:
## Reproduce 

* Load a code (with `function onInit(){...`)into ESP(12)
* Toggle power 
* See your code runs as normal. 
* Run `E.setBootCode("console.log('this is bootcode');", true)`
* Toggle power 
* See constant resets: 
    > ets Jan  8 2013,rst cause:2, boot mode:(3,0)
    > 
    > load 0x40100000, len 1396, room 16 
    > tail 4
    > chksum 0x89
    > load 0x3ffe8000, len 776, room 4 
    > tail 4
    > chksum 0xe8
    > load 0x3ffe8308, len 540, room 4 
    > tail 8
    > chksum 0xc0
    > csum 0xc0
    > 
    > 2nd boot version : 1.4(b1)
    >   SPI Speed      : 80MHz
    >   SPI Mode       : QIO
    >   SPI Flash Size & Map: 32Mbit(512KB+512KB)
    > jump to run user1 @ 1000

* Reflash firmware 
* Load an example code
  
   ```
onInit = function(){console.log("heyyyyyyyyyy");}
save()
   ```

* Toggle power 
* See last code runs as normal at boot.

tve, 2016-08-07:
The reason is the call to strlen at https://github.com/espruino/Espruino/blob/master/src/jsparse.c#L2524 which causes non-word accesses to the string in flash.

gfwilliams, 2016-08-09:
Ahh, good spot - thanks :)

Well, that is a bit of a hack, but in the scheme of things it's not that bad... Adding some kind of `AS_FLASH_ADDR` macro to https://github.com/espruino/Espruino/blob/master/src/jsutils.h#L58  might be cool, as you could re-use it where you do a similar thing with 0x40200000 in jshardware.c

But to be honest it's fine - some kind of comment might be nice though  :)

MaBecker, 2016-12-09:
@ceremcem - can this be closed ?

MaBecker, 2017-01-07:
@gfwilliams - fix works - please close this issue

closed 2017-01-09 by gfwilliams