espruino/Espruino #894: Multiple 'then' in Promises

gfwilliams, 2016-08-07:
Looks like there's a problem:

http://forum.espruino.com/conversations/290967/

Can't believe I didn't spot/test this at the time, but should be an easy fix.

closed 2016-09-20 by gfwilliams

gfwilliams, 2018-07-05:
Thanks, I created an issue for it ^^^

Basically this is when you call `then` on a promise that has already been resolved. I didn't even realize that was allowed.

If you want to work around it, just add all the `then` and `catch` you need to it before you resolve it.

Did you actually hit this in real-life code, or were you running through some tests?

radist2s, 2018-07-05:
Yes, I use Promises in my code. But I can imagine how difficult and costly is to make realization of Promise in Espruino(utilization and other). I can deal without Promises of course, but It will be more verbose code. I suppose, if promises can't be realized natively it should be removed and if some one will need it Promises can be written on JS directly(can't imagine how costly on controller it will be). Promises is really important thing in JS, for example to make `async/await` functions, which transpiled by Babel really useful when you doing any asynchronous code.

gfwilliams, 2018-07-05:
If you use a new Espruino build from travis, those issues that you found should now be fixed. They'll be sorted in 1v100 anyway

commit a62cfe50 (2016-09-20): finally solve promise chaining, incl what happens with rejection chaining
Fix 'chained' promises (fix #894)