espruino/Espruino #895: Support Realtek RTL8711AF, RTL8711AM, RTL8195AM, RTL8710

Ti-webdev, 2016-08-10:
Package: QFN-48 (6 x 6 mm)
CPU: ARM Cortex M3 @ 166MHz
RAM: 48KiB available for user firmware
Flash: 1MiB on-chip
WiFi: 802.11 g/n (up to 150 Mbps) 1x1
GPIO: Up to 21 GPIO pins
I²C: Up to 3 I²C ports
PCM: Up to 2
PWM: Up to 4 PWM ports
UART: 1 high speed, 1 low speed
Power requirements: 80mA @ 3.0 to 3.6V
Temperature range: -40°C to + 125°C
Certifications (not yet confirmed): FCC/CE/TELEC/SRRC/WiFi Alliance

gfwilliams, 2016-08-15:
Any word on pricing?

The problem I have is that the ESP8266 is pretty close in this area, is dirt cheap, and already has a good port of Espruino. I guess the question is what would be gained by us spending months porting to this board?

drasko, 2016-08-24:
@Ti-webdev you are talking about this one: http://www.cnx-software.com/2016/08/18/getting-started-with-bt-rtl-00-rtl8710-module-serial-console-at-commands-and-esp8266-pin-to-pin-compatibility/?

daanpape, 2016-08-25:
I have also been thinking about espruino on the RTL8710. I think it's very feasible, there's much information on the RTL8710forum (https://www.rtl8710forum.com). I don't think it's months of work as it's just an ARM Cortex M3.

tve, 2016-09-17:
>  I don't think it's months of work as it's just an ARM Cortex M3.
The processor is not the issue, the wifi stack is. To run Wifi you pretty much need some RTOS and Espruino has not been ported to any. Does the RTL8710 use the GCC toolchain? Does it use FreeRTOS? Is the Wifi driver open source or how is it integrated?

MaBecker, 2017-02-14:
https://www.seeedstudio.com/RTL8710-WiFi-Module-p-2793.html?utm_source=mailchimp&utm_medium=edm&utm_campaign=new_0214

closed 2017-12-15 by gfwilliams