espruino/Espruino #896: MICROBIT.py pins don't match edge connector i/o pins

veryalien, 2016-08-14:
The espruino microbit pin definitions don't match the numbering given here:
[https://www.microbit.co.uk/device/pins]

It first appeared that only the Espruino docs were wrong, but it now looks like D3/P3 (the leftmost pin on the edge connector) got forgotten. D3 is LED COL 1 not D4.
It is very confusing that the espruino numbering does not match the edge connector breakout board pin numbers.

gfwilliams, 2016-08-14:
Ok, so you're saying that Button A is actually on D5 and not D4, because it says so on https://www.microbit.co.uk/device/pins ?

Have you tested this, or are you just comparing diagrams?

Because I just tested with  a micro:bit here:

* Connect LED between 3v and pin D4
* Press BTN A
* LED lights

So for me at least, http://www.espruino.com/MicroBit#pinout  matches the two micro:bits I own.

Now they were pre-production units but it still seems highly unlikely that they'd be different.

gfwilliams, 2016-08-14:
Please can you test on your board. Both with `digitalRead(D4)`, `digitalRead(D5)`, and an LED between 3v and D4.

veryalien, 2016-08-14:
I've already checked this with a MicroBit edge connector from Kitronik [https://www.kitronik.co.uk/pdf/5601_edge_connector_breakout_board_for_the_bbc_microbit_datasheet_v1_2.pdf].
Pin 5 on the edge connector, not pin 4) reacts to a button press (inverted as expected, it's normally high).
The edge connector is clearly labelled 3, 0, 4, 5 and it's the fourth pin from the left on the micro bit that changes state with a button press, hence "pin 5".
They took pins 0, 1 and 2 and made the first three large contacts and then just put the other pins in between, starting at 3, along the edge connector.
I think (maybe I'm wrong) that your D4 is reading the signal of D5. I think your D4 definition is wrong, you've missed D3 in devices, so it's not in step with the other definitions.
As far as I can see Espruino is consistent, but ut doesn't match all the other evidence from other sites.
I really wish board designers could number pins clearly, consistently and logically. Usually the microcontroller ports themselves are sensibly labelled. To me the Pi is a similar 'random mess'!

veryalien, 2016-08-14:
It depends how you are defining D4, both logically and physically. I'm using the numbering I see on all the reference sites and measuring the pins directly off the microbit (although the edge connector is also effectively wired straight-through).
Pins directly on the Microbit from left to right with the LEDs facing you (just to be pedantic):
3 - small
0 - large
4 - small
5 - small - this reacts when I press BTNA
6- small
7- small
1 - large
8 - small
9 - small
10 - small
11 - small - BTNB
2 - large

I don't know if it's a between the ears problem, or if something has changed at some point.

gfwilliams, 2016-08-14:
Where did you get your micro:bit from? And recently? I have two, from two different stages of micro:bit development, and both of them are the same - Pin 4 - as shown in your diagram - is connected to the button, and not pin 5.

It's not an issue with me 'missing' D3, as you'll almost certainly find that D3 works, and BTNB is also in the correct place.

Perhaps the BBC have changed the micro:bit pinout for some random reason - but it's frustrating if so - especially after handing out micro:bits for development and then not informing developers there was a change in the production version!

It's also pretty odd that you're the first one to complain, since I'd imagine testing BTN1 is the *first* thing any micro:bit user would do.

Does the display work for you? (via `show(...)`) Or is that broken too?

veryalien, 2016-08-14:
Everything works. LEDs and both buttons BTN1 and BTN2. Is is just a naming issue?
I only noticed that there seems to be a labelling difference when I tried connecting through the edge connector to light LEDs from both buttons BTN2 on pin 11 works but, again, it's the fourth pin on the edge connector board that toggles off/on when I press/release Button A.
I was trying the third pin labelled "4" for ages, until I realized that the pinout diagram shows it on the next pin along. They label it "5", on the pinout diagram it is labelled as P5.
I think it might have been changed at some point, look what I just found, see the "schematic" aka pinout at the bottom of this page: [https://thecustomizewindows.com/2016/03/bbc-micro-bit-specs-schematic/]
**Oh look Button A on P4. At some point it must have been changed!!**
My MicroBit came from Pimoroni this week together with the Kitronik edge connector in the prototyping system, the edge connector board seems to be labelled V1.1, not sure if that is relevant or not.
Why does BTN1 work on my microbit, the physical pins don't seem to match?

veryalien, 2016-08-14:
I've got the suspicion that you are doing everything right in the code and I'm doing everything right with the edge connector.
The signal that you read from BTN1/D4 is coming out on P5 on the edge connector - not directly from the physical button like I was led to believe. Does that, at least, make some illogical sense?

veryalien, 2016-08-14:
I've tested P0-P16 from the edge connector connecting each pin in turn through a switch connected to GND when pressed. I did digitalRead(D0) to D16.
The pins D4 and D5, according to your definitions vs P4 and P5 on the edge connector, **are reversed**.
When I connect my switch to P4 and do a digitialRead(D4) or digitalRead(BTN1) it stays at 1, no matter what the switch state.
When I'm connected to P4 (third pin on the edge connector) and I do digitalRead(D5) the state changes (inverted) according to the switch connected to P4.
When I'm connected to P5 (fourth pin on the edge connector) and I do digitalRead(D4) or digitalRead(BTN1) the state changes (inverted) according to the switch connected to P5.
So, on my microbit that affects reading the state of BTN1 (which reacts to P5) and doesn't match your D4, which is why I raised the issue in the first place.
Doing digitalRead BTN2/D11 with a switch on P11 works ok.

veryalien, 2016-08-14:
And then to make it all totally clear, you could read this confusing description of pin 5 (or do they mean pin 4), again from [https://www.microbit.co.uk/device/pins]
**pin 5**: GPIO shared with Button A. This lets you trigger or detect a button "A" click externally. This pin has a pull-up resistor, which means that by default it is at voltage of 3V. To replace button A on the BBC micro:bit with an external button, connect one end of the external button to **pin 4** and the other end to GND. When the button is pressed, the voltage on **pin 4** is pulled down to 0, which generates a button click event.

gfwilliams, 2016-08-15:
So as far as I can tell then, your button is *still* connected to P4 on the microcontroller itself.

I guess it's possible that for some unknown reason they have swapped P4 and P5 on the board's header, *while keeping the actual connections the same*. I'll ask around and find out.

gfwilliams, 2016-08-15:
Ok, I just looked up and it turns out that P4/5/etc bear no resemblance to the 'real' microcontroller pins, so swapping them over won't actually be too difficult to do. I'm just waiting to hear back on whether anything else needs swapping

gfwilliams, 2016-08-15:
And yes, I just heard back. They did change the connector pinout.

So now all I have to do is update to the correct mapping (when I find it).

commit d7e03a8a (2016-08-15): Fix pin header numbering for BBC micro:bit (it changed for the production version) (fix #896)
closed 2016-08-15 by gfwilliams