espruino/Espruino #898: Can't put script into flash area under specific circumstances

yerpj, 2016-08-16:
Hi !

I recently tried to download a ~300 lines long script and save it to flash area on a Nucleo (411RE) with Espruino 1v86. The script seems to be correctly downloaded into RAM (it takes a few tens of seconds) but after that the flash don't seem to be written. Instead of having the debug message telling that the flash is being written, I have the message that says something like "can't get prompt".

With small scripts, no problems, everything works as expected.

did anyone ever came across this issue?

gfwilliams, 2016-08-16:
There are a few questions about similar things [in the forum](http://forum.espruino.com/).

Usually it happens when your code actually runs stuff in the main scope (so as it is uploaded), rather than just defining functions and variables.

For instance `while (1);` will cause this behaviour, because it just causes an infinite loop.

Also, even if your code just takes a while to execute, it can cause problems since there is no flow control and the input buffer can fill up and drop characters. Does the Web IDE actually manage to recover back to the console? If so, type `E.getErrorFlags()` to see if the buffer got full up.

yerpj, 2016-08-16:
Thanks Gordon, I will read the forum about potential same behaviours.

closed 2016-08-16 by yerpj