espruino/Espruino #900: Serial6 object is now defined on Nucleo 401RE

yerpj, 2016-08-17:
Hi,
Does anyone ever tried to use the Serial6 on Nucleo401?
It seems odd , cause it is referenced on the board pinout down here: http://www.espruino.com/ReferenceNUCLEOF401RE

yerpj, 2016-08-22:
I just had a try on Nucleo411, using auto-completion feature to list the available serial ports. It appears that we have only access to Serial1 to 3. But when typing Serial3.setup(), the interpreter does not seem to find the equivalent function .See below:
``` 
>Serial
Serial1             Serial2
Serial3
>Serial3.setup()
Uncaught InternalError: setDeviceClockCmd: Unknown Device 800
 at line 1 col 15
Serial3.setup()
              ^
```

I had a look at source code but I could figure out how and where the build process is creating board-dependant ressources

gfwilliams, 2016-08-23:
Looks from here like there is no USART3 on the chip at all: https://github.com/espruino/Espruino/blob/master/targetlibs/stm32f4/lib/stm32f411xe.h

really in the boards file it should either report the uart count as 2, or as 6. But if it's 6 then you're going to have similar issues with uarts 3 to 5

yerpj, 2016-08-23:
You're right but if I look at the [datasheet](http://www.st.com/content/ccc/resource/technical/document/datasheet/b3/a5/46/3b/b4/e5/4c/85/DM00115249.pdf/files/DM00115249.pdf/jcr:content/translations/en.DM00115249.pdf) at page 44, I see that there is a USART6, that can be mapped on PA11 and PA12, exactly as stated in the [Espruino board reference page](http://www.espruino.com/ReferenceNUCLEOF401RE). I expected be able to use it.
I will try to modify the [board config](https://github.com/espruino/Espruino/blob/master/boards/NUCLEOF411RE.py) with
 ``` 
'usart' : 6,
```
instead of 3 and figure out if I can access the Serial6 object.

gfwilliams, 2016-08-23:
I'm afraid it's never been a priority since I'm producing builds for the Nucleo F411 for free.

If I was, I might be more motivated to fix bugs :)

yerpj, 2016-08-23:
I totally agree with you, and I am using the Pico as much as I can, but for one application, I need to use 3 serial ports and I could not achieve this with the Pico (unless the softwareSerial is available yet, but I don't think so). It is the reason I tried with Nucleo, until I figured out that one of the 3 serial ports was the native console :-/

I don't know much about exposing hardware ressources with Espruino but I will try to make Serial6 usable anyway.

gfwilliams, 2016-08-23:
:) yeah, just upping the 'usart' figure to 6 might well do it. It just means there will be Serial3/4/5 shown that don't work.

closed 2020-12-09 by gfwilliams