espruino/Espruino #901: MAC Address Problem

iremkorkmaz, 2016-08-22:
with my code wifi.on('probe_recv')
macid's I receive are wrong, because I guess it was a copy+paste code from other triggered events.
how can I fix this?

gfwilliams, 2016-08-23:
I'm pretty sure we need more information. What board is this? can  you provide example code? what's the actual + expected result?

iremkorkmaz, 2016-08-23:
**We are using Adafruit Huzzah Feather.**

**We flash the chip with v1.86 from the link**
http://www.espruino.com/binaries/git/commits/master/

**and with the command :** 
python ./esptool.py --port=/dev/cu.wchusbserial1410  write_flash  -fm=dio -fs=32m 0x00000 “flash-file-name”

**in our code :** 
wifi.on('probe_recv', function(data) {
        console.log(data);
});

**We have a data logged here but the MAC ID that we received is different from the device's MAC ID.** 

**By checking out the code here from the library:** 
https://github.com/espruino/Espruino/blob/b0292fc5aef46a09f51d8cac85e8e5a746e3e6d8/libs/network/esp8266/jswrap_esp8266_network.c

**When we connect the device to esp8266 and get its MAC ID, it is correct, but when we don't connect the device and ping ESP 8266 we get its MAC ID wrong.**

**So   case EVENT_SOFTAPMODE_STACONNECTED: is working fine for us , but   case EVENT_SOFTAPMODE_PROBEREQRECVED: is not working fine.**

gfwilliams, 2016-08-23:
Thanks! @tve, one for you?

Looks like here https://github.com/espruino/Espruino/blob/b0292fc5aef46a09f51d8cac85e8e5a746e3e6d8/libs/network/esp8266/jswrap_esp8266_network.c#L2004

`mac = evt->event_info.sta_connected.mac;` should be `mac = evt->event_info.ap_probereqrecved.mac;`?

tve, 2016-08-24:
yup!

iremkorkmaz, 2016-08-25:
When there will be an update ? :)

This function is one of our core items.  So if you can give me an update or how can we fix that on our side?

All the best!

iremkorkmaz, 2016-08-31:
@tve , could you please give an update about this?

tve, 2016-08-31:
I confirmed that I agree with the change suggested by gordon. I've wanted to fix this in the repo but this week-end was too busy. It's a small change but it will take me some time to actually test it, so it's not a 5 minute operation for me. It's not like you're paying me to make the change for you either, right? Since all this is open source it is trivial for you to make the change yourself, so I don't understand why you feel that you are blocked. You could even fix it, test it, and create a pull-request. That's certainly the fastest way to get fixes included...

gfwilliams, 2016-09-02:
As @tve says - you can fix things yourself or could even pay us to make changes - but we have absolutely no obligation to fix problems you're having for free - that's not how Open Source works.

If you want something a bit better supported, you can also buy one of the boards that I sell - since I actually make money from those I'm a bit more motivated to fix bugs.

closed 2016-09-20 by gfwilliams