espruino/Espruino #902: Date() as function call returns a Date object but should return a string

luwar, 2016-08-22:
`Date()` should return a string equals to `new Date().toString()`.

Current implementation:
```
>Date()
={ "ms": 32880814.10312652587 }
>Date() instanceof Date
=true
```

Expected result:
```
>Date()
="Thu Jan 1 1970 09:10:41 GMT+0000"
>typeof Date()
="string"
```
See https://stackoverflow.com/questions/9584719/date-vs-new-date-in-javascript

gfwilliams, 2016-08-23:
Ergh. That's horrible.

Thanks for reporting - but this one's going to be quite far down my priority list I'm afraid...

gfwilliams, 2024-10-10:
Closing this now - nobody seems to have been concerned about this in the last 8 years, and looking at BangleApps there are a whole bunch of people using `Date().something` whose code would be broken if I fixed this