espruino/Espruino #906: Long event names on ESP8266

the1laz, 2016-08-28:
I don't seem to be able to fire events with names longer than 8 characters. obj.on('presentation') won't fire if I obj.emit('presentation'), but I can call it with obj['#onpresentation']()

I tried shortening the event name until I got to 7 characters, which then worked.

This only seems to be a problem on ESP8266. There was a forum post about long event names a little while ago, not sure if this is related. http://forum.espruino.com/conversations/277524/

the1laz, 2016-08-29:
I've now found this on an stm32 as well, looks like something I'm doing might be causing this. I'll do some more investigation, but I'm not really sure how to debug it.

gfwilliams, 2016-09-05:
You'd probably want to check up what's in memory using `trace` - it could be to do with the variable name being long enough that 2 variables are needed for it. If you can reproduce it on a normal Linux compile it'd help for debugging (but on those you could be looking at a significantly longer event name to trigger it)

the1laz, 2016-09-06:
Thanks, I'll give that a go.

tve, 2016-09-17:
Given "I've now found this on an stm32 as well" does the esp8266 tag make sense on this issue?

gfwilliams, 2016-09-17:
oops. Good point :)

commit 7a8c690b (2016-09-20): Fix emitting of events with long names (fix #906)
closed 2016-09-20 by gfwilliams