espruino/Espruino #910: save() reports -4 bytes available on nRF51

ddm, 2016-08-31:
This behaviour has been tested with the following nRF51 variants (QFN48):

- nRF51822-QFAC (256k Flash + 32k RAM) using NRF51822DK=1 with a PCA10028 DK
- nRF51822-QFAA (256k Flash + 16k RAM) using MICROBIT=1 and NRF51TAG=1 using a custom board equivalent to the micro::bit

To reproduce using the nRF51822 development kit (PCA10028) using a QFN48 nRF51422-QFAC SoC (equivalent to nRF51822 but with ANT capabilities):

1. `git clone https://github.com/espruino/Espruino.git && cd Espruino`
2. `git co 1317aa0` (optional, that is the HEAD I tested)
3. `NRF51822DK=1 RELEASE=1 make`
4. Flash `espruino_1v86.1614_nrf51822.hex` to your PCA10028 DK

Calling `save() `, regardless of the size of the running code (e.g. the stock "blink" example), yields `ERROR: Too big to save to flash (xxx vs -4 bytes)`.

universal, 2016-09-01:
I think i had a similar issue playing with a waveshare clone, and still have other issues. I think i fixed the byte issue with a changed config, might have just been pages in the chip section which i changed. I hope i don't forget it tomorrow, i only have the code at my work machine since it is still toying around :-)

ddm, 2016-09-01:
Funny you mention that: I noticed it works on my custom board (but not the PCA10028) when I just add `-g3` compiler flag. No amount of work on the `saved_code` board config values seem to do any good though.

gfwilliams, 2016-09-02:
On micro:bit this works I think. Change `pages:0` to `pages:3` in your file -  seems to be a bug in the nRF51DK board file

closed 2016-09-02 by gfwilliams

ddm, 2016-09-02:
@gfwilliams Thank you for the fix Gordon! I just realised that changing `gcc` flags had nothing to do with me getting it to work: I had set `pages:3` on my custom board too... I guess the nRFTag board should be updated too since it uses the same nRF51822-QFAA variant as the micro:bit and my custom board.

commit 852b4ef4 (2016-09-02): Fix #910