espruino/Espruino #912: Esp8266 - Makefile needs firmware updated from 1.4 to 1.5

wilberforce, 2016-09-06:


wilberforce, 2016-09-15:
@tve 

Please can you add the 1.5 bootloader referenced here:
https://github.com/espruino/Espruino/blob/master/.travis.yml#L20

esp_iot_sdk_v1.5.0 (1).tgx 

esp_iot_sdk_v1.5.0 (1)/esp_iot_sdk_v1.5.0/bin/

The Makefile also needs a tweak:
github.com/espruino/Espruino/blo­b/master/Makefile#L1927

BOOTLOADER = "$(ESP8266_SDK_ROOT)/bin/boot_v1.4(b1).b­in"
to
BOOTLOADER = "$(ESP8266_SDK_ROOT)/bin/boot_v1.5.bin"

gfwilliams, 2016-09-15:
And also the ESP8266 readme

wilberforce, 2016-09-16:
@tve 
I saw that the http://s3.voneicken.com/esp_iot_sdk_v1.5.4.tgx has the `boot_v1.5.bin`

Are we ready to go to SDK 1.5.4?

https://github.com/espruino/Espruino/pull/924

wilberforce, 2016-09-19:
The has been superseded by a sdk 2.0 build

closed 2016-09-19 by wilberforce