espruino/Espruino #914: ESP8266 - PWM Freq Flicker (Most likely still broken PWM Code)

Hypfer, 2016-09-08:
I'm trying to PWM an LED Strip running 1v86.171 using

> pinMode(D5, "output");
> analogWrite(D5, 0.3, {freq: 30000});

It flickers terribly even though it shouldn't since its 30000hz. Setting the freq value higher makes it light up at full brightness constantly. The freq param works since it does work well with low hz values like 1-100hz

gfwilliams, 2016-09-08:
I'm afraid this is something that's never going to work on ESP8266. There is no hardware PWM peripheral on ESP8266 so PWM is done with software.

Realistically you'll never be able to do anything much above 1000 Hz, and even then it might be a bit rough. It could produce an exception for any frequency above 1000 I guess, but to be honest that might just annoy a lot of people and you could still make it glitchy by adding loads of PWMs.

If you want proper peripherals, and lots of them (SPI, I2C, PWM, ADC, etc) then you'll need to use a board with a separate Microcontroller. The Espruino boards will do it, and there should be a WiFi Espruino board coming in a month or two as well.

gfwilliams, 2016-09-08:
Yes, you could write some C code to receive data and do stuff and could connect via serial - it could be a bit of a pain though.

Or you could use a PCA9685 - after all, it's designed to drive LEDs. Adafruit even do one on a board.

Or... simplest might actually be to get an Espruino Pico, flash your ESP8266 with the AT command firmware, connect it up, and access WiFi that way.

MaBecker, 2017-08-13:
For ESP8266 there is a replacement for the SDK PWM code:

https://github.com/StefanBruens/ESP8266_new_pwm

Maybe someone whats to jump in and do the replacement.

Comment from @gfwilliams (#1175)
> Yeah, so what you'd want to do for ESP8266 is look in https://github.com/StefanBruens/ESP8266_new_pwm/blob/master/pwm.c, pull out all the stuff to do with pwm_intr_handler and the timer registers, and shoehorn that same code into the util_timer stubs in jshardware.c... Then ESP8266 PWM/waveform/etc would be loads better

stefan-sherwood, 2017-11-12:
I had this same flickering problem driving an LED strip using PWM on the ESP8266 with MicroPython. 

I tried disabling that Access Point interface and the problem is 100% resolved for me. The "station" interface is still active and I am able to run a web server on the board while doing PWM and it works fine, flicker-free.

The MicroPython code for this is:

```
import network
ap_if = network.WLAN( network.AP_IF )
ap_if.active( False )
```

MaBecker, 2017-12-02:
@stefan-sherwood  please share your analogWrite statement

stefan-sherwood, 2017-12-02:
```
from machine import Pin
pin = Pin( 16, Pin.OUT, Pin.PULL_UP )
pin.freq( 1023 )

brightness_percentage = 30
pin.duty( brightness_percentage / 100 * 1023 )
```

I am not changing the clock frequency because it works fine as is, but you can double the default 80 MHz ESP8266 clock.

```
import machine
machine.freq( 160000000 ) # double default frequency of 80 MHz
```

Sorry about using MicroPython examples but that's what I'm using.

MaBecker, 2018-07-03:
Seem there is no interest after six month to implement this,  so lets close this.

closed 2018-07-03 by MaBecker