espruino/Espruino #923: Array.map(String.fromCharCode) not working as intended

ceremcem, 2016-09-16:
I'm not able to understand the problem is because of the `Array.map` function or the `String.fromCharCode` function, because following codes work as intended: 

```
>E.toUint8Array([99, 100, 101]).map(function(x){return 2*x})
=new Uint8Array([198, 200, 202])
>String.fromCharCode(99)
="c"
```

...but these don't: 

```
>[99, 100, 101].map(String.fromCharCode)
=[ 
  "c\x00\x00", 
  "d\x01\x00", 
  "e\x02\x00"
 ]

// expected `[c, d, e]`

>[99].map(String.fromCharCode)
=[ 
  "c\x00c"
 ]
// expected `['c']`

>E.toUint8Array([99, 100, 101]).map(String.fromCharCode).join('')
="000"   
// expected "cde"
```

gfwilliams, 2016-09-17:
This doesn't work on a desktop browser either - you should really check on those first.

It's a common problem - array.map supplies **more than one argument**, so fromCharCode creates a string based on all arguments given to it, and you get >1 character.

closed 2016-09-17 by gfwilliams

ceremcem, 2016-09-17:
Oh, sorry about that. I was so sure about the code that it should work as this way :(

gfwilliams, 2016-09-17:
I know, but that's JavaScript for you :) `.map` especially is a real pain sometimes because of those extra arguments.

I added `E.toString` to convert arrays straight to strings if that's what you're after? It's much more efficient

ceremcem, 2016-09-17:
Arrgghh!! How did I miss `E.toString` all this time? Yes, it's what I tried to do... Thank you again... :)