espruino/Espruino #929: ERROR: JshPinState #defines have changed, please update pinStateToString()

MaBecker, 2016-09-26:
Message during boot on ESP board:

`````
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v87.1711 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>rll|là|lìb|ìrbbònnlnnâìbplrlrlpònàlbnâ|bònnîll`nnl`nrnl`pònàrâàbnâ|ìbònnîl`nnl`nrnlpònàrâàìlbnâ|bònnîl`nnl`nrnl`rnl`rlnònnârnnnlblìlìlll~rlìlllnâllllìlnànìn~llàbìllpbbrìl`nlpìl`bl`lnnânl|bll`rlâìnìròn|ìll`ârlþ
ERROR: JshPinState #defines have changed, please update pinStateToString()
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v87.1711 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
> 

````

"VERSION": "1v87.1711",
  "BUILD_DATE": "Sep 26 2016",
  "BUILD_TIME": "20:13:17",
  "GIT_COMMIT": "351e779cdcd997753720d31f2a8e41a6ae8b221d",

build options:

RELEASE=1,EMBEDDED=1, USE_NET=1, USE_TELNET=1,USE_GRAPHICS=1, and SDK 2.0

and dump() after reboot does not show any pinMode settings anymore,

MaBecker, 2016-09-26:
seems to be something complete went wrong with my build and flashed image ......

gfwilliams, 2016-10-18:
`make clean`?

MaBecker, 2016-10-18:
a reboot() lead to a multiple  resets and starts without pinMode() 

````

783945> Espruino resetting the esp8266
=undefined
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 1856, room 16
tail 0
chksum 0x63
load 0x3ffe8000, len 776, room 8
tail 0
chksum 0x02
load 0x3ffe8310, len 552, room 8
tail 0
chksum 0x79
csum 0x79
2nd boot version : 1.5
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user2 @ 81000
âìnìròn|ìll`ârlþERROR: JshPinState #defines have changed, please update pinStateToString()

......

 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v87.1711 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>dump();
=undefined
````

MaBecker, 2016-10-18:
a reset(); returns with default  pinmode settings

````
>reset();
=undefined
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v87.1711 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>dump();
pinMode(D0, "input_pullup");
pinMode(D4, "input_pullup");
pinMode(D5, "input_pullup");
pinMode(D12, "input_pullup");
pinMode(D13, "input_pullup");
pinMode(D14, "input_pullup");
pinMode(D15, "input_pullup");
=undefined
> 

````

MaBecker, 2016-10-18:
what is missing ?

MaBecker, 2016-10-18:
@gfwilliams yes - make clean does not fix this - please a a ESP8266 label

MaBecker, 2016-10-18:
@tve  any idea ?

tve, 2016-10-18:
Are you trying to compile master? Have you made changes? I've never seen this but I also haven't updated by fork to master since the 1.87 release. I'd need more info on how to repro.

MaBecker, 2016-10-18:
hmm - just did a new clone, flashed, serial connect and power up - same behavior 

````
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v87.1773 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>rll|là|lìb|ìrbbònnlnnâìbpìlrlrlpònàlbnâ|bònnîll`nnl`nrnl`pònàrâàbnâ|ìbònnîl`nnl`nrnlpònàrâàìlbnâ|bònnîl`nnl`nrnl`rnl`rlnònnârnnnlblìlìlll~rlìlllnâllllìlnànìn~llàbìllpbbrìl`nlpìl`bl`lnnânl|bll`rlâìnìròn|ìll`ârlþ
ERROR: JshPinState #defines have changed, please update pinStateToString()
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v87.1773 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>dump();
=undefined
> 
````

````
 >reset();
=undefi
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v87.1773 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>dump();
pinMode(D0, "input_pullup", true);
pinMode(D4, "input_pullup", true);
pinMode(D5, "input_pullup", true);
pinMode(D12, "input_pullup", true);
pinMode(D13, "input_pullup", true);
pinMode(D14, "input_pullup", true);
pinMode(D15, "input_pullup", true);
=undefined
> 
````

MaBecker, 2016-10-19:
more infos about build:

````
>process.env
={
  "VERSION": "1v87.1711",
  "BUILD_DATE": "Sep 26 2016",
  "BUILD_TIME": "20:13:17",
  "GIT_COMMIT": "351e779cdcd997753720d31f2a8e41a6ae8b221d",
  "BOARD": "ESP8266_BOARD",
  "CHIP": "ESP8266",
  "CHIP_FAMILY": "ESP8266",
  "FLASH": 0, "RAM": 81920,
  "SERIAL": "5ccf7fc1-120b",
  "CONSOLE": "Serial1",
  "EXPORTS": { "jsvLock": 1075850212, "jsvLockAgainSafe": 1075850268, "jsvUnLock": 1075850644, "jsvSkipName": 1075858932,
    "jsvMathsOp": 1075864108, "jsvMathsOpSkipNames": 1075865796, "jsvNewFromFloat": 1075856892, "jsvNewFromInteger": 1075856792, "jsvNewFromString": 1075856448,
    "jsvNewFromBool": 1075856860, "jsvGetFloat": 1075862004, "jsvGetInteger": 1075858684, "jsvGetBool": 1075862332, "jspeiFindInScopes": 1075876968,
    "jspReplaceWith": 1075876688, "jspeFunctionCall": 1075886960, "jspGetNamedVariable": 1075878776, "jspGetNamedField": 1075879220, "jspGetVarNamedField": 1075879352,
    "jsvNewWithFlags": 1075856188 }
 }
>require("ESP8266").getState()
={
  "sdkVersion": "2.0.0(5a875ba)",
  "cpuFrequency": 160, "freeHeap": 9360, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
````

MaBecker, 2016-10-19:
Question: why is  require("ESP8266).reboot() and reset() different in terms of pinMode ?

MaBecker, 2016-11-20:
flashed 1v89 but still get this error on powerup or reboot ......

````
>rll|là|lìb|ìrbbònnlnnâìbplrlrlpònàlbnâ|bònnîll`nnl`nrnl`pònàrâàbnâ|ìbònnîl`nnl`nrnlpònàrâàìlbnâ|bònnîl`nnl`nrnl`rnl`rlnònnârnnnlblìlìlll~rlìlllnâllllìlnànìn~llàbìllpbbrìl`nlpìl`bl`lnnânl|bbll`rlâìnìròn|ìll`ârlþ
ERROR: JshPinState #defines have changed, please update pinStateToString()
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v89.2035 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
> 

>require("ESP8266").getState();
={
  "sdkVersion": "2.0.0(5a875ba)",
  "cpuFrequency": 160, "freeHeap": 8560, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
> 
````

can someone please explain what this is about ?

clowrey, 2016-11-20:
What version of GCC are you using?

MaBecker, 2016-11-20:
xtensa-lx106-elf-gcc-4.8.2

clowrey, 2016-11-20:
When flashing did you fully erase the chip? And does it work with the online compiled binaries that Espruino Travis creates?

MaBecker, 2016-11-20:
no fully erase  - using wiflash.sh

same result i flashing with travis build  espruino_1v89.7_esp8266.tgz

clowrey, 2016-11-20:
Does it work with past version of espruino or none? I would try a few from
various times. If it works with none of the precomputed binaries then there
is some problem with the hardware you're using or method of flashing. I
know
That PuckJS is the main new development and there could be
incompatibilities in these newer builds with other hardware, which will be
fixed in time if that's the case..

MaBecker, 2016-11-20:
fully erase and flashed with 1v85, 1v86, 1v87 - see the results below 

1v85.138 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
-- manual reboot --
>rll|là|lìb|ìrbbònnlnnâìbplrlrlpònàlbnâ|lìlbònnîll`nnlnrnll`pònàrâàbnâ|ìbònnîl`nnlnrnl`pònàrâàìbnâ|lbònnîl`nnl`nrnrl`rnrl`rlnònnârnnnlbbplblìlìlll~rlìlllnâllllìlnànìn~llàbìllpbbrìl`nlpìl`bl`lnnânl|bll`rldon't use rtc mem data
r

1v86.185 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
-- manual reboot --
>rll|là|lìb|ìrbbònnlnnâìbpìlrlrlpònàlbnâ|lìlbònnîll`nnlnrnll`pònàrâàbnâ|ìbònnîl`nnlnrnl`pònàrâàìbnâ|lbònnîl`nnl`nrnrl`rnrl`rlnònnârnnnlbbplblìlìlll~rlìlllnâllllìlnànìn~llàbìllpbbrìl`nlpìl`bl`lnnânl|bll`rldon't use rtc mem data
r

1v87 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
-- manual reboot --
>rll|là|lìb|ìrbbònnlnnâìbplblrlpònàlbnâ|bònnîll`nnl`nrnl`pònàrâàbnâ|ìbònnîl`nnl`nrnlpònàrâàìlbnâ|bònnîl`nnl`nrnl`rnl`rlnònnârnnnlblìlìlll~rlìlllnâllllìlnànìn~llàbìllpbbrìl`nlpìl`bl`lnnânl|bll`rlâìnìròn|ìll`ârlþERROR: JshPinState #defines have changed, please update pinStateToString()

clowrey, 2016-11-20:
I think that first block of gibberish is from the esp8266 chip bootloader core that comes out at a different baud rate than the 9600 Espruino uses, I forget if it's 115200 or some other, you can look it up though.

When I have flashed ESP8266 chips in the past I have had similar issues, I think what ended up working for me was using the node MCU flasher tool GUI and then flashing with my espruino code after.

gfwilliams, 2016-11-24:
Just search for pinStateToString: https://github.com/espruino/Espruino/blob/6da2ddd05917d98e0693a084a2d64fcfb8b800d1/targets/esp8266/jshardware.c#L304

it's trying to convert the enum defined here: https://github.com/espruino/Espruino/blob/master/src/jshardware.h#L102

into strings - but it's just an array, so if the enums change then it'll get upset. Just copy the enum names into that array in order and you'll be sorted.

wilberforce, 2016-11-29:
@MaBecker 

This has been added:
  JSHPINSTATE_GPIO_OUT_OPENDRAIN_PULLUP, ///< GPIO pin as open-collector/open-drain output WITH PULLUP

So:
```
/**
 * Convert a pin state to a string representation.
 * This is used during debugging to log a meaningful value instead of a
 * numeric that would then just have to be decoded.
 */
static char *pinStateToString(JshPinState state) {
  static char *states[] = {
    "UNDEFINED", "GPIO_OUT", "GPIO_OUT_OPENDRAIN",
    "GPIO_IN", "GPIO_IN_PULLUP", "GPIO_IN_PULLDOWN",
    "ADC_IN", "AF_OUT", "AF_OUT_OPENDRAIN",
    "USART_IN", "USART_OUT", "DAC_OUT", "I2C",
  };
  return states[state];
}
```

Add:

```
/**
 * Convert a pin state to a string representation.
 * This is used during debugging to log a meaningful value instead of a
 * numeric that would then just have to be decoded.
 */
static char *pinStateToString(JshPinState state) {
  static char *states[] = {
    "UNDEFINED", "GPIO_OUT", "GPIO_OUT_OPENDRAIN",
   "GPIO_OUT_OPENDRAIN_PULLUP",
    "GPIO_IN", "GPIO_IN_PULLUP", "GPIO_IN_PULLDOWN",
    "ADC_IN", "AF_OUT", "AF_OUT_OPENDRAIN",
    "USART_IN", "USART_OUT", "DAC_OUT", "I2C",
  };
  return states[state];
}
```

MaBecker, 2016-11-29:
@wilberforce  - thanks - I will test and create a pull request in the next days

MaBecker, 2016-12-08:
created pull request #992

avanc, 2016-12-16:
I have the same problem with the precompiled binaries (1v90) from the Espruino website.

     ets Jan  8 2013,rst cause:1, boot mode:(3,7)
    
    load 0x40100000, len 2408, room 16 
    tail 8
    chksum 0xe5
    load 0x3ffe8000, len 776, room 0 
    tail 8
    chksum 0x84
    load 0x3ffe8310, len 632, room 0 
    tail 8
    chksum 0xd8
    csum 0xd8
    
    2nd boot version : 1.6
      SPI Speed      : 80MHz
      SPI Mode       : QIO
      SPI Flash Size & Map: 32Mbit(512KB+512KB)
    jump to run user1 @ 1000
    
    ����n�r��n|�l�l`��r�l��ERROR: JshPinState #defines have changed, please update pinStateToString()

MaBecker, 2016-12-16:
@avanc  - check the pull request #992

avanc, 2016-12-16:
@MaBecker : Thanks for the hint. Will the pull request go upstream? Or is it a speacility of the ESP8266?

avanc, 2016-12-16:
I started to compile Espruino on my own (SDK including cross-compiler, compiling espruino) but I got stuck while compiling Espruino:

    CC libs/network/esp8266/network_esp8266.o
    libs/network/esp8266/network_esp8266.c: In function 'doClose':
    libs/network/esp8266/network_esp8266.c:466:5: error: implicit declaration of function 'espconn_abort' [-Werror=implicit-function-declaration]
         espconn_abort(pSocketData->pEspconn);
         ^
    libs/network/esp8266/network_esp8266.c: At top level:
    cc1: warning: unrecognized command line option "-Wno-float-conversion" [enabled by default]
    cc1: warning: unrecognized command line option "-Wno-discarded-qualifiers" [enabled by default]
    cc1: some warnings being treated as errors
    make: *** [Makefile:1877: libs/network/esp8266/network_esp8266.o] Fehler 1

@MaBecker : Would you mind to share your binaries?

olliephillips, 2016-12-17:
I'm seeing a similar error with Onewire and DS18B20. After only a few cycles board resets and console offers the same error message "ERROR: JshPinState #defines have changed, please update pinStateToString()". Using 1.89 firmware on ESP8266.

MaBecker, 2016-12-17:
try this:

https://github.com/MaBecker/esp8266/tree/master/firmware/espruino_1v89.2053_esp8266

MaBecker, 2016-12-17:
let me know if this works for you

alcir, 2016-12-18:
esptool.py --port /dev/ttyUSB0 --baud 115200   write_flash --flash_freq 80m --flash_mode qio --flash_size 32m 0x0000 "boot_v1.6.bin" 0x1000 espruino_esp8266_user1.bin   0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin

So your espruino_1v89.2053_esp8266 seems to be ok.

I get the same issue as @avanc using espruino_1v90_esp8266 downloaded from espruino site.

avanc, 2016-12-18:
The patched version 1v89.2053 works on my board.

@MaBecker : Thanks, you saved my weekend.

avanc, 2016-12-18:
It seems that I have to dig a little bit deeper what the patch is doing.
I using the following board: http://cgi.ebay.de/ws/eBayISAPI.dll?ViewItem&item=291850429865

If I'm not wrong I need GPIO5 to activate the relay. However, GPIO5 is always low (I tried digitalWrite(D5,true) ). On the other hand, GPIO12 works as expected.

I now saw that something changed with GPIO5 using the patch. So I'm curious what the patch is doing.

MaBecker, 2016-12-18:
@avanc there is a sanity check for pin function enum to catch ordering changes

the patch just reacts to the change of the enums - thats all

just check yourself https://github.com/espruino/Espruino/pull/992/commits/8b0f7895913d0d2350fad59c5568b2da47241dbc

MaBecker, 2016-12-18:
Hi @tve  PR https://github.com/espruino/Espruino/pull/992/commits/8b0f7895913d0d2350fad59c5568b2da47241dbc does not stop 1v90 from rebooting with cause 2 

 Any hints  how analyse such a issue  ?

MaBecker, 2016-12-20:
got it working with a new clone, sdk 1.5.x and 2.0.x for 1v89 and 1v90

closed 2016-12-20 by MaBecker