espruino/Espruino #930: Impossible to form two DIFFERENT synchronized waveforms on two outputs.

VAGrishin, 2016-09-27:
1. This program should to form two different waveforms - sine and cosine.
In other words phase shift between two outputs should be equal 90 degrees.

`var w = new Waveform(128);
var w2 = new Waveform(128);
for (var i=0;i<128;i++) w.buffer[i] = 128+Math.sin(i*Math.PI/64)*127;
for (var i=0;i<128;i++) w2.buffer[i] = 128+Math.cos(i*Math.PI/64)*127;
analogWrite(A4, 0.5);
analogWrite(A5, 0.5);
var t = getTime()+0.01; // start in 10ms
w.startOutput(A4, 4000, {repeat:true, time:t});
w2.startOutput(A5, 4000, {repeat:true, time:t});`

Unfortunately the phase shift between two outputs is equal to zero.
The same signal is present on the two pins. The signal is limited sine wave.

2. DAC of the microcontroller can operate in 12-bit mode.
Unfortunately only 8-bit mode is supported by the Espruino.

Controller: STM32F405RG
Board: http://amperka.ru/product/iskra-js 
    ____       __
   /  _/_____ / /__ _____ ____ _
   / / / ___// //_// ___// __ `/
 _/ / (__  )/ ,<  / /   / /_/ /
/___//____//_/|_|/_/    \__,_/
Based on Espruino 1v85.1113
(c) 2016 G.Williams, Amperka LLC

gfwilliams, 2016-09-27:
Thanks for the report! Unfortunately I won't be able to help you as I don't support those boards.

Amperka might be able to help you if you contact them though?

VAGrishin, 2016-09-27:
I have reported the problem to Amperka a few days ago.
They have confirmed the existence of the problem.
But then there was no any reaction.
They did not even answer to the question of who support this board.

gfwilliams, 2016-09-27:
Yeah, it's a pain - if a user of one of the Espruino boards I sell was having the problem I'd fix it, but Amperka don't contribute anything to the Espruino project when they sell a board.

I have enough trouble making enough money to support my work on Espruino - I can't afford to support another company's users for free!

WebReflection, 2016-09-27:
Not the first time I read these kind of requests and answers, I think you should change [this section](https://github.com/espruino/Espruino#current-state) naming it **Support State** instead, and also put such session higher in the page (or link it on the top).

Maybe in doing so, people would stop before asking (or maybe a list of well known **unsupported boards** could help too).

gfwilliams, 2016-09-27:
Good idea - I wonder if that many people would read it, but it's a start. It might also encourage some of the companies that use Espruino to put some of their profits back into helping me support their users.

Thing is: this is a useful bug report, and may apply to Espruino boards as well - so I'm definitely leaving it open. If someone in the community wants to fix it, or [donate something](http://www.espruino.com/Download) so that I can spend time tracking it down and fixing it, that'd be awesome.

VAGrishin, 2016-09-28:
> Yeah, it's a pain - if a user of one of the Espruino boards I sell was having the problem I'd fix it, but Amperka don't contribute anything to the Espruino project when they sell a board.
>
> I have [...]

I understood everything.
I'll try to see the source code at https://github.com/espruino/Espruino

The controller STM32F405RG and the board seems to work properly.
Function analogWrite allows to form two different waveforms - sine and cosine.
However, the clock frequency of waveforms is too low.

closed 2016-09-28 by VAGrishin

gfwilliams, 2016-09-28:
Turns out this is an issue on Espruino boards as well so I've just pushed a fix. You'd have to do your own build for the IskraJS board though.

VAGrishin, 2016-09-29:
Thank you very much. 

I am familiar with the process of application building under Linux. 
My primary operating system is Debian.

commit ee834eac (2016-09-28): Fix regression with 2 concurrent waveforms on different pins (fix #930)