espruino/Espruino #936: NRF.setServices must clear out existing services

gfwilliams, 2016-10-04:
Currently services just get added onto the end each time it's called - even after reset, which is definitely not what's needed.

ddm, 2016-10-06:
Ideally we would like to reset the attribute table at the beginning of `NRF.setServices` to make the call idempotent. The UART service would have to be reset if affected.
Have you found a way to reset the SoftDevice ?

A temporary fix might be to only add services and characteristics that have not been defined yet (using the same loop as `NRF.updateServices`). It would not allow to change existing ones but it would prevent having the same one defined multiple times.

gfwilliams, 2016-10-06:
I just asked about this here: https://devzone.nordicsemi.com/question/97828/removing-gatt-services/

But as you suggest I'll tweak the code to re-use your UUID->handle code now

gfwilliams, 2016-10-06:
Quick question about your changes - you had a #define for the maximum value length - is that actually needed? There are no buffers allocated and if it were too big, the softdevice would just error rather than silently dropping chars.

ddm, 2016-10-06:
Correct. The current use of the `MIN` macro silently drops values instead of failing. I haven't checked if an 'overflow' error code exists and if it is properly handled though...

Maybe removing the `APP_CFG_CHAR_MAX_LEN` constant and the use of `MIN` would be better indeed.

ddm, 2016-10-06:
Furthermore I should have used the declared `maxLen` instead of a constant to have if consistent with `NRF.setServices` so my current solution is confusing at best in case the updated value length is too big.

gfwilliams, 2016-10-06:
Ok, done. setServices is still not ideal since it seems we can't actually get back from service UUID->handle. Still, it's a stop-gap until I figure out how to fix things properly.

ddm, 2016-10-06:
The last modifications seem to have changed the behaviour of `NRF.updateServices`. I now get a different `err_code` on line 1700 of [jswrap_bluetooth.c](https://github.com/espruino/Espruino/blob/master/libs/bluetooth/jswrap_bluetooth.c#L1700) using the  same JS code as earlier today. I'll try to debug the issue.

gfwilliams, 2016-10-06:
Argh, bummer. Try now.

I tested it, but not with notifications. Sorry about that.

ddm, 2016-10-06:
No worries. It might just be an `NRF_ERROR_DATA_SIZE` because my previous implementation allowed me to be careless about the packet sizes...

ddm, 2016-10-06:
Yes it's fixed now. Thanks!

gfwilliams, 2016-10-07:
Hopefully properly fixed (allowing characteristic removal) with: https://github.com/espruino/Espruino/commit/6212f76fda800ce06c48969d266349da10f8c521

As it happens I had to store the services list anyway, so I guess there might be the option of integrating it and the BLE client stuff so you can do:

```
var b  = NRF.setServices({ 0x1234 : { 0x2345 : { value : "foo" }});
b[0x1234][0x2345].write("bar");
```

I'm not sure how worth it that is... It might allow some more advanced usage though?

closed 2016-10-07 by gfwilliams

ddm, 2016-10-07:
This is seriously cool! I don't reconfigure services that often now that I have settled on the services and characteristics I need but having the option makes rapid development so much simpler (especially when starting on a project).

I like the `write` call personally. I will probably still use `NRF.updateServices` when I want to update several values and use `notify: true`.

ddm, 2016-10-07:
And thank you for updating the SDK ;)

gfwilliams, 2016-10-07:
No problem - the new SDK does break the bootloader though, so I need to get that sorted :(

The real pain was when you `reset()` - when uploading code you really want to start from the same point each time, so it just would have been pretty frustrating otherwise. It would have been a lot harder to get it working if it wasn't for the folks on Nordic Devzone though :)