espruino/Espruino #937: Softdevices are missing for nRF5

ddm, 2016-10-07:
Currently `NRF52832DK=1 make` fails with
```
IOError: [Errno 2] No such file or directory: 'Espruino/targetlibs/nrf5x/nrf5_sdk/components/softdevice/s132/hex/s132_nrf52_3.0.0_softdevice.hex'
make: *** [espruino_1v87.1738_nrf52832.hex] Error 1
make: *** Waiting for unfinished jobs....
```
Same thing with the `NRF51822DK` where `s130_nrf51_2.0.1_softdevice.hex` is missing.

I don't know whether the SoftDevice can or should be part of the Espruino repo. Maybe the README could explain where to get them and how to install them.

ddm, 2016-10-07:
The link to the s132 is https://www.nordicsemi.com/eng/nordic/download_resource/56261/5/22102858

gfwilliams, 2016-10-07:
Ahh, sorry - try that. I re-arrange things a little as well to remove a level from the directory structure.

The .gitignore file had been stopping me seeing the hex files :)

closed 2016-10-07 by gfwilliams

ddm, 2016-10-07:
I just downloaded them but I was wondering if redistribution by Espruino was authorised.

gfwilliams, 2016-10-07:
The original hex files were placed in the repo by a Nordic employee, so I'm just going to assume they are. Of course if Nordic complain I will take them out, but for now it's easier for everyone :)