espruino/Espruino #938: GATT service configuration now delayed over BLE UART

ddm, 2016-10-07:
If I understand correctly, since the recent changes to how GATT services are configured for nRF5 devices, it is not possible to call `NRF.setServices` and the `NRF.updateServices` over a BLE/UART connection (cf. [jswrap_bluetooth.c](https://github.com/espruino/Espruino/blob/master/libs/bluetooth/jswrap_bluetooth.c#L1497) line 1497).

In my experience it works fine to update the GATT attribute table over a BLE/UART connection. Would it be possible to force the update if "I'm feeling lucky" ?

ddm, 2016-10-07:
It can be very useful to update the running code from a SmartPhone using just the BLE UART service (e.g. no USB-serial converter available).
For example, the following works fine if we remove the delayed GATT update:
```
NRF.setServices({
  0xff20: {
    0xff21: {
      value: 0,
      maxLen: 1,
      readable: true,
      notify: true
    }
  }
});
setInterval(function () {
  NRF.updateServices({
    0xff20: {
      0xff21: {
        value: Math.round(100 * analogRead(D3)),
        notify: true
      }
    }
  });
}, 1000);
```

It is probably only safe as long as the UART service (and its RX and TX characteristics) is untouched...

gfwilliams, 2016-10-07:
Yeah, it'd be a simple to tweak it.

I reckon it's fine as long as:

* You're only adding stuff
* You're not touching UART (which you can't at the moment anyway)

ddm, 2016-10-07:
My thoughts exactly. I was just wondering if you had run into problems using `setServices` over BLE/UART...

gfwilliams, 2016-10-07:
Not really, no - it was more laziness when I did it - just wanted to make sure everything was working and commit :)

gfwilliams, 2016-10-14:
Ok, I think this is fixed now. Basically if you're just adding services the first time around then you're fine - it's only if you call it again that it wants to restart

closed 2016-10-14 by gfwilliams