espruino/Espruino #939: JS support differences between board

ddm, 2016-10-07:
When compiling Espruino for the `NRF52832DK`, the following code runs fine:
```
Array(10).fill(0)
```

However, using the  `NRF51822DK` it throws an error:
```
Uncaught Error: Function "fill" not found!
```

Is the JS support dependant on available resources or MCU model?

gfwilliams, 2016-10-07:
Yeah, it has to use the SAVE_ON_FLASH define, and that turns off a few extras. Array.fill is ES 2015 I think so that gets dropped.

gfwilliams, 2016-10-13:
Closing this - I think that's as expected.

The documentation even covers it: http://www.espruino.com/Reference#l_Array_fill

`Note: This is only available in some devices: not devices with low flash memory`

closed 2016-10-13 by gfwilliams