espruino/Espruino #941: Console characters not echoing properly

nkolban, 2016-10-10:
When we connect the WebIDE to the ESP32 environment and enter text, it does not echo back immediately.  Only when we press enter do we see the entered text and its execution.  Investigation needs to be done on how to properly integrate the handling of input characters and having them echoed.

gfwilliams, 2016-10-10:
I imagine this is to do with how FreeRTOS handles characters. In the Linux build there's some still that initialises the input stream to 'raw' mode - you might need something similar

nkolban, 2016-10-10:
Thanks Gordon.  If you are interested in FreeRTOS, this page might be a useful study.

http://www.freertos.org/FreeRTOS_Features.html

Despite having the words "Operating System" in its name, it is not at all similar to an operating system like Linux or Windows.  Specifically, FreeRTOS provides:

1) Memory management routines (like malloc/free)
2) Task management and scheduling
3) Semaphores, locks, mutexes, queues and lists
4) ... from an ESP32 perspective ... nothing else

There is no concept of device drivers, file systems, shells, commands or tools.  FreeRTOS is a library of functions rather than any kind of a "kernel".

In the port so far of Espruino to the ESP32, I don't think there is one utilization of FreeRTOS.

That all said ... these are very, very early days in everyones study of ESP32 and our understanding of the UART interface as exposed by Espressif is still up for grabs.

gfwilliams, 2016-10-10:
I do know a bit about FreeRTOS - it seemed like it might have provided some simple printf/readline kind of commands that might have been causing you problems. Obviously not though.

Just FYI I see you put `What if we want to use the serial for data input?` at: https://github.com/espruino/Espruino/blob/ESP32/targets/esp32/jshardware.c#L101

Even doing what you're doing, you can still use that Serial device for data input - the queue handles all IO, not just for Espruino's console. All you need to do is (as you suggest) call `jshPushIOCharEvents` from an IRQ instead and you're completely sorted.

nkolban, 2016-10-11:
After some digging ... the fix was found to be pleasantly easy.  In the first pass, we were using:

```
printf("%c", c);
```
to write the character to the output.  Changing this to write directly to uart using:
```
uart_tx_one_char(c);
```
immediately cured up the issue.  It looks like `printf()` was likely buffering the output.

closed 2016-10-11 by nkolban