espruino/Espruino #948: Maximum number of scopes exceeded

gfwilliams, 2016-10-14:
With callbacks hell it's actually pretty easy:

```
function go() {
 setTimeout(function() {
  setTimeout(function() {
   setTimeout(function() {
    setTimeout(function() {
     setTimeout(function() {
      setTimeout(function() {
       setTimeout(function() {
        setTimeout(function() {
         setTimeout(function() {
          result = 1;
         }, 10);
        }, 10);
       }, 10);
      }, 10);
     }, 10);
    }, 10);
   }, 10);
  }, 10);
 }, 10);
}

go();
```

Ideally we should use a JS array for scopes to avoid the hard limit

vshymanskyy, 2018-02-28:
Ok, I was actually able to get `async/await` working on Espruino (using Babel transpiler and some hacking). **Here is a full example:  https://github.com/vshymanskyy/espruino-await**

The uglify module sometimes inlines and chains functions and Promises (with some minification options) and I have started facing this issue. Had to disable some optimizations.

gfwilliams, 2019-03-25:
It's been hit again here: http://forum.espruino.com/conversations/332111/#comment14671765

gfwilliams, 2019-03-26:
Just to add, I have a workable solution to this which doesn't impose any limits. Just debugging some memory leaks and it'll be sorted

closed 2019-03-27 by gfwilliams

zored, 2020-06-07:
Is there any way to increase this limit? I went too far with my async/await calls and no there is no way back 😂

gfwilliams, 2020-06-08:
How are you using async/await? A transpiler?

You could build your own firmware with this changed I believe? https://github.com/espruino/Espruino/blob/e78f33f03065f7e15e207c67f7deb279241cf269/src/jsutils.h#L252

zored, 2020-06-09:
@gfwilliams thank you a lot for this reference! Yes, I use `fast-async` babel transformer mentioned above.

Generally I'm transpiling TypeScript. Despite some callbacks overhead it [works great](https://github.com/zored/arduino/blob/5f3a16d30e3a61aa0c5a45dc9b55008df42f81ec/sketch/ts/wifi/index.ts#L14).

opichals, 2020-06-09:
@gfwilliams Maybe I am missing something but it looks like this limitation should have been dropped since Mar 2019.

https://github.com/espruino/Espruino/search?q=JSPARSE_MAX_SCOPES doesn't find any use of the JSPARSE_MAX_SCOPES define.

gfwilliams, 2020-06-09:
That's a very good point - I'll remove that define.

@zored looks like you're just using a really out of date firmware then?

zored, 2020-06-09:
@gfwilliams That's right! We have a board called [Iksra JS](http://wiki.amperka.ru/js:iskra_js) in Russia and [its firmware](https://github.com/amperka/Espruino/tree/iskrajs) is based on some really outdated version of Espruino. I will try to keep it up to date.

commit 5734a947 (2019-03-27): Remove limit on the number of scopes that can be searched (fix #948)