espruino/Espruino #960: Add support for HTTPS to ESP32

nkolban, 2016-10-25:
Add support for HTTPS to ESP32 board.  This issue will track research and development for this capability.

gfwilliams, 2016-10-26:
It should 'just happen' if you've implemented the networking API.

HTTPS support in Espruino sits on top of the socket API, so it works on any platform (assuming there is the RAM/ROM for it).

Just use the correct `USE_XXX` defines in the Makefile and it'll work. You'll also get crypto functionality too.

I'd advise against using the Espressif provided implementation - it'll be a lot of work to drag it in, and will probably involve a whole load of `ifdef ESP32` all over the HTTP and networking codebase

nkolban, 2016-10-27:
Exactly as you said, it worked 1st time without ANY hooking in of https libraries from ESP32 land.  ESP32 has hardware assist for SSL/TLS but these require the use of the ESP32 native libraries.  However, since all is working, there is zero imperative to leverage these at this time.  The sample program that I used to test was:

```
// Connect to an access point and get a web page.
//
var ssid="guest";
var password="kolbanguest";

var wifi=require("Wifi");
wifi.connect(ssid, {password: password}, function() {
  console.log("Connected to access point, getting web page");
  var http = require("http");
  http.get("https://httpbin.org/ip", function(res) {
    res.on('data', function(data) {
      console.log(data);
    });
  });
});
```

closed 2016-10-27 by nkolban

gfwilliams, 2016-10-27:
Great! Yeah, I guess at some point we could look at abstracting the support out, but for now it should be fine for pretty much everyone.