espruino/Espruino #969: esp32-hal-spi.h is not added ?

lapsule, 2016-11-10:
Got this on ESP32 target branch:

> 
> targets/esp32/jshardware.c: At top level:
> targets/esp32/jshardware.c:45:27: fatal error: esp32-hal-spi.h: No such file or directory
>  #include "esp32-hal-spi.h"
>                            ^
> compilation terminated.
> make: *** [targets/esp32/jshardware.o] Error 1
> make: *** Waiting for unfinished jobs....
>

wilberforce, 2016-12-05:
check your paths - I think that file is part of the arduino libs that are getting pulled for SPI/i2c

This stuff is really in flux at the moment - the v1.0 of the esp-idf has just been released and trying to get that up and running now

wilberforce, 2017-02-08:
@gfwilliams 

I think this can be safely closed as we no longer are using the arduino libs..

gfwilliams, 2017-02-08:
Ok, great - thanks!

closed 2017-02-08 by gfwilliams