espruino/Espruino #973: nRF5x: Trying to updateServices with notify:true can reset the board

gfwilliams, 2016-11-11:
```
NRF.setServices({"0001":{"2A56":{"readable":true}}});

function go() {
  NRF.updateServices({
    "0001" : {
      "2A56" : {
        value : 1,
        notify: true
      }
    }
  });
 }
```

If `notify:true` is set in setSevices then it's all fine

closed 2016-11-14 by gfwilliams

UDXS, 2018-07-30:
I'd like to report that this still happens on the BBC Microbit (1v99)

gfwilliams, 2018-07-30:
Does any error message get displayed on the Serial output?

Also, does it work as expected when a) not notofying, b) the service actually had notifications enabled

UDXS, 2018-07-30:
Actually, it might be an issue with my computer specifically (maybe the Windows Bluetooth impl.). The Micro:bit simply resets when I attempt to connect and, of course, the connection fails. So it's not happening with notify (though, in my code, I have it notify even if there is no connection). This may be a memory/implementation issue as my computer works perfectly fine with the Puck.js.

gfwilliams, 2018-07-30:
Yes - it'll probably be to do with Pairing. To connect, Windows needs to pair (which requires encryption keys to be stored and things - which the Micro:bit has no room for). Connecting to a phone app will probably work fine.

commit c969f3c5 (2016-11-14): nRF5x: Make sure that updateServices(notify) on a non-notifyable service will just error, rather than reset (fix #973)