espruino/Espruino #978: enhancement for drawString - add a font size multiplier

MaBecker, 2016-11-23:
Normally I use the vectorfont to draw large strings on oleds.

Using oled-i2c-bus  on beaglebone brings a nice feature that allows a font size multiplier as parameter for  [writeString](https://github.com/baltazorr/oled-i2c-bus#writestring)

This is how it is done [Oled.prototype._drawChar](https://github.com/baltazorr/oled-i2c-bus/blob/master/oled.js#L215) - not sure how complex this will be - but like to try.

Benefit: use one fixed font in multiple size like 1:4x6, 2:8x12, 3:12x18 ......

Thought this could be a smart enhancement to have.

@gfwilliams  What do you think ?

gfwilliams, 2016-11-23:
Have you used `Graphics.setFontVector`? That gives you an actual vector font that can be resized however you want, with no pixellation. I guess it's quite possible that you're using ESP8266 and the font has been removed to save some space in flash though?

I'd try using that first and see how it goes?

MaBecker, 2016-11-23:
Ups - have not stated that clear - Yes on Espruino I use the setFontVector - sure what else ;-)

Just a thought ..... I will do some testing and come back with results or close this issue.

gfwilliams, 2016-11-23:
The actual draw code is here: https://github.com/espruino/Espruino/blob/master/libs/graphics/jswrap_graphics.c#L625

There are 3 different paths though:

* built-in 4x6 font
* vector font
* custom font

The easiest option would be to add a way to set custom fonts with a scale factor - I'm not sure how much use that would be.

Maybe the most useful enhancement would be to add a scaling option to `drawImage`. That way you could draw whatever you wanted onto a small 'Graphics' instance and resize it?

gfwilliams, 2017-03-17:
Is it fine to close this now? With the 4MB build you'll be able to include vector fonts, so will be able to resize properly

MaBecker, 2017-03-17:
Yes  I close

closed 2017-03-17 by MaBecker