espruino/Espruino #984: Strange behavior with objects when using string keys from numbers.

Hacker-CB, 2016-12-01:
**1. Seems that it is bug when I'm using string which contains only numbers as object key:**

Code example:
```
var obj={};
obj["1233456789012345"]="yes";
```

Will return next object:
`{  "-1983824007": "yes" }`

**2. Second example of the strange behaviour:**
Code example:
```
var obj={};
obj["1233456789012345"]=true;
```

Will return next object:
`{ "true": true }`

I found this problems because I'm operating with 1-wire bus and I tried to use 1-wire rom address as key.

gfwilliams, 2016-12-02:
Thanks! It sounds like the issue is that the key value is too big to fit into 32 bits. My guess is that many of the key values will actually work, but some like the one you found won't - it should be a relatively easy fix in Espruino.

For now, you should be able to force it to use strings by adding something to the value to turn it into a string:

```
var obj={};
obj["x"+"1233456789012345"]="yes";
obj["x"+onewire.address] = ...;
```

Hopefully that won't be too difficult to modify

Hacker-CB, 2016-12-02:
Yes, I have already fixed that by adding prefix in JS, but it is not look very nice. 
I will wait for global fix.
Thank you.

commit d8c05a37 (2016-12-15): Fix bug when using >32 bit integers as array indices (fix #984)
closed 2016-12-15 by gfwilliams