espruino/Espruino #996: nRF5x: allow advertising of services

gfwilliams, 2016-12-14:
Currently it's only UART and HID, but easy enough to change via advertising_init

OwenBrotherwood, 2017-02-20:
[Battery Status](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.battery_service.xml)

http://forum.espruino.com/conversations/300527/#comment13482090 with edited initial comment  to take into account Gordons feedback

gfwilliams, 2017-02-20:
Sorry - I don't understand.

How is this different from the example right at the top of setAdvertising: http://www.espruino.com/Reference#l_NRF_setAdvertising

Where battery status is reported? You can do this now. It works fine.

OwenBrotherwood, 2017-02-20:
Sorry - I thought it was an open issue ie that there was no support (allow advertising of services)
I haven't had a chance to investigate if possible: currently investigating an issue in node-gyp https://github.com/nodejs/node-gyp  / issues / 1116 (I don't want to link direct as it gives a issue link that is not needed in this context

Is this a closed issue?

gfwilliams, 2017-02-21:
There's a difference between the data sent in an advertising packet and the list of services in an advertising packet.

Sending battery level data is fine, but saying, for instance - 'this device implements a heart rate monitor interface' is different - and that's what this bug was about.

commit bd39d2cc (2017-03-15): nRF5x: Allow services to be advertised (fix #996)
closed 2017-03-15 by gfwilliams