espruino/Espruino #1007: ES6 syntax doesn't work on micro:bit

RX14, 2016-12-25:
I recently got a BBC micro:bit, and have installed Espruino 1v90 on it. It's working great, and i'm enjoying being ale to use a JS REPL on a microcontroller.

The changelog says that ES6 arrow functions and template literals were introduced in 1v88. However attempting to use template literals or arrow functions results in a syntax error:
```
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v90 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>let foo = `bar`;
Uncaught SyntaxError: Got TEMPLATE LITERAL expected EOF
 at line 1 col 11
let foo = `bar`;
          ^
>let bar = (() => "foo");
Uncaught SyntaxError: Got ')' expected EOF
 at line 1 col 13
let bar = (() => "foo");
            ^
>bar();
Uncaught Error: Function "bar" not found!
 at line 1 col 1
bar();
^
```

The template literals especially seem to be an interesting error message as it appears to me the lexer knows of the "template literal" token, yet the parser seems to fail parsing it.

I'm not entirely sure if let/const lex, but they seem to be treated differently to variables at least:
```
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v90 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>var;
=undefined
>let;
=undefined
>foo;
Uncaught ReferenceError: "foo" is not defined
>const;
=undefined
>conss;
Uncaught ReferenceError: "conss" is not defined
```

gfwilliams, 2017-01-04:
Hi - yes, I'm afraid this is something that had to be left out of the Micro:bit due to space constraints.

On these devices the Bluetooth takes up around 100kB of memory, so on Puck.js Espruino as around 400kB to use, but on the micro:bit it's only 150kB. Unfortunately something has to give.

If this had been mentioned on the micro:bit page, would it have helped?

RX14, 2017-01-04:
Yes, mentioning this in the docs would be appreciated.

closed 2017-01-04 by gfwilliams