espruino/Espruino #1015: raw esp8266-12e - how to point Waveform to the analog pin

mrbar42, 2017-01-07:
Hi,

I use esp8266 boards directly with espruino v1.89 (no devboard).
As i read i many other [issues](https://github.com/espruino/Espruino/issues/929) and forum discussions
analogRead() ignores the pin and uses the only available one.
But does Waveform acts the same?
I couldn't get any data defining any pin number to Waveform.

In a more general fashion i wonder if it would be helpful to add the pins to the `ESP8266` module similar to `NodeMCU` making usage more clear and explicit:
```js
var esp8266 = require('ESP8266');
esp8266.A0
esp8266.D0
```

MaBecker, 2017-01-22:
for PWM check #914

gfwilliams, 2017-01-23:
I'm afraid waveform isn't really supported on ESP8266 at the moment.

As far as I know, ESP8266 Waveform doesn't properly use interrupts anyway (so it gets stalled by JS execution). As a result, even if this could be made to work, you'd still be a very glitchy and unreliable waveform read out of it.

mrbar42, 2017-01-25:
Then there's no issue here I guess.

Thanks.

closed 2017-01-25 by mrbar42