espruino/Espruino #1023: Allow Puck.js to drive Neopixel/WS281x/APA10x RGB pixels

gfwilliams, 2017-01-13:
This might be solved with the addition of: https://github.com/espruino/Espruino/issues/695

But we need to test.

MaBecker, 2017-01-21:
Hi Gordon,

tried this simple code to control some neopixels with a Puck, expected to have eight white leds,  but did not work. I guess there is something wrong with the code I am using.

```
var LEDs = 8*3;
var SPI1 = new SPI();
SPI1.setup({baud:3200000, mosi:D1});
var rgb = new Uint8ClampedArray(LEDs).fill(25);
SPI1.send({data:rgb,count:LEDs});
```

MaBecker, 2017-01-21:
Oh completely forgot about that  NPs need special timing like used with

```
SPI2.send4bit([255,0,0], 0b0001, 0b0011); 
```

Can  the ESP8266.neopixelWrite()  be implemented as NRF.neopixelWrite()  to handle this ?

MaBecker, 2017-01-22:
Have a workaround for now: Puck - USART - ESP8266 - Neopixel 

Also started to port ESP8266.neopixelWrite() to NRF.neopixelWrite().

* stuck with the asm stuff 

* what are  the equivalent calls  for:

     GPIO_REG_WRITE(GPIO_OUT_W1TS_ADDRESS, pinMask); 
     ets_intr_lock()
     ets_intr_unlock()

could need some help .... 

[jswrap_puck.h.txt](https://github.com/espruino/Espruino/files/721885/jswrap_puck.h.txt)
[jswrap_puck.c.txt](https://github.com/espruino/Espruino/files/721886/jswrap_puck.c.txt)

gfwilliams, 2017-01-23:
It's probably better to do as suggested in this bug, and to just enable the full-speed SPI write - in which case the code from your 2nd post should work.

Disabling interrupts on nRF52 can be a bit iffy, as the radio gets upset (I think there's a watchdog timer).

DMA SPI would be perfect, but there's a big in the nRF52's hardware which means that single-byte SPI writes fail via DMA, so to get it working I'll have to make a special SPI function that falls back to non-DMA for single byte transfers :(

MaBecker, 2017-01-23:
Ok, I understand - Dropped the idea of implementing NRF.neopixelWrite()

gfwilliams, 2017-02-24:
Just FYI - I have this done now - just waiting to check on licensing of the code I'm using before I commit it.

I'll be adding `E.neopixelWrite` which'll work the same on all platforms.

MaBecker, 2017-02-24:
Wow - this is so cool -  Thanks  !

MaBecker, 2017-02-24:
@gfwilliams Is there something you want me to change for ESPs  ?

gfwilliams, 2017-02-24:
Nope, it should be fine - I moved the code around so `E.neopixelWrite` is added for it too, and `ESP8266.neopixelWrite` just forwards to `E.neopixelWrite` for compatibility.

When I commit it'd be great if you could test though, just to make sure I didn't break anything :)

MaBecker, 2017-02-24:
sure,  I can test

closed 2017-02-27 by gfwilliams

MaBecker, 2017-02-27:
ran some test - works perfect  - thanks !

gfwilliams, 2017-02-28:
Great - thanks!

MaBecker, 2017-03-01:
One more point to add, this will cause confusion:

> Write to NeoPixel/WS281x/APA10x-style LEDs attached to the given pin

this function can not handle  [APA10X](http://dangerousprototypes.com/blog/wp-content/media/2014/09/apa102.jpg), because they need two data lines.

[APA's need SPI](http://forum.espruino.com/conversations/280762/?offset=25#comment12997609) :)

gfwilliams, 2017-03-01:
It's unfortunate - I guess I'll have to spell out the part numbers individually, APA104 are fine: https://groups.google.com/forum/#!topic/atxhs-discuss/r_bGlXBZC-E - as are some other APAs as well

commit 002938f4 (2017-02-27): Add E.neopixelWrite for all platforms (fix #1023)