espruino/Espruino #1031: MICROBIT error: unable to find a register to spill

OwenBrotherwood, 2017-01-23:
Using the Espruino Vagrantfile, I can create an nRF52-DK but am failing with micro:bit.
Any hints?

```
vagrant@vagrant-ubuntu-trusty-64:~/Espruino$ MICROBIT=1 RELEASE=1 make
LD espruino_1v91.18_microbit.elf
targets/nrf5x/bluetooth.c: In function 'jsble_set_services':
targets/nrf5x/bluetooth.c:1246:1: error: unable to find a register to spill
 }
 ^
targets/nrf5x/bluetooth.c:1246:1: error: this is the insn:
(insn 931 1842 938 88 (parallel [
            (set (mem:SI (reg/f:SI 1006 [739]) [0  S4 A32])
                (mem:SI (reg/f:SI 1050 [738]) [0  S4 A32]))
            (set (mem:SI (plus:SI (reg/f:SI 1006 [739])
                        (const_int 4 [0x4])) [0  S4 A32])
                (mem:SI (plus:SI (reg/f:SI 1050 [738])
                        (const_int 4 [0x4])) [0  S4 A32]))
            (set (reg/f:SI 1006 [739])
                (plus:SI (reg/f:SI 1006 [739])
                    (const_int 8 [0x8])))
            (set (reg/f:SI 1050 [738])
                (plus:SI (reg/f:SI 1050 [738])
                    (const_int 8 [0x8])))
            (clobber (reg:SI 957))
            (clobber (reg:SI 958))
        ]) targets/nrf5x/bluetooth_utils.c:178 775 {movmem8b}
     (expr_list:REG_UNUSED (reg/f:SI 1050 [738])
        (expr_list:REG_UNUSED (reg/f:SI 1006 [739])
            (expr_list:REG_UNUSED (reg:SI 958)
                (expr_list:REG_UNUSED (reg:SI 957)
                    (nil))))))
targets/nrf5x/bluetooth.c:1246: confused by earlier errors, bailing out
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [espruino_1v91.18_microbit.elf] Error 1

```

gfwilliams, 2017-01-23:
That sounds like your compiler's broken.

Did you try just following the specific nRF51/52 instructions? https://github.com/espruino/Espruino/blob/master/README_Building.md#for-nordic-semiconductors-nrf51nrf52-series-devices - specifically the compiler mentioned there.

OwenBrotherwood, 2017-01-23:
Yes: read and found the CI site as well, where micro:bit is compiling with the compiler that is used there (creating hex that are a bit larger than necessary)
NRF52-DK is working: flashed to board.
MICROBIT has the error using the Vagrantfile
https://github.com/espruino/Espruino/blob/master/Vagrantfile

OwenBrotherwood, 2017-01-23:
I double check compiler

OwenBrotherwood, 2017-01-23:
```
vagrant@vagrant-ubuntu-trusty-64:~/Espruino/scripts$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

OwenBrotherwood, 2017-01-23:
USE_BLUETOOTH= 0 was a hope for the code for any bluetooth would not be compiled, but it did and created an error

```
else ifdef MICROBIT
EMBEDDED=1
SAVE_ON_FLASH=1
# Save on flash, but we still want the debugger and tab complete
DEFINES+=-DUSE_DEBUGGER -DUSE_TAB_COMPLETE
BOARD=MICROBIT
OPTIMIZEFLAGS+=-Os
USE_BLUETOOTH=0
USE_GRAPHICS=1

```
```
vagrant@vagrant-ubuntu-trusty-64:~/Espruino$ MICROBIT=1 RELEASE=1 make
LD espruino_1v91.18_microbit.elf
targets/nrf5x/bluetooth.c: In function 'jsble_set_services':
targets/nrf5x/bluetooth.c:1246:1: error: unable to find a register to spill
 }
 ^
targets/nrf5x/bluetooth.c:1246:1: error: this is the insn:
(insn 931 1842 938 88 (parallel [
            (set (mem:SI (reg/f:SI 1006 [739]) [0  S4 A32])
                (mem:SI (reg/f:SI 1050 [738]) [0  S4 A32]))
            (set (mem:SI (plus:SI (reg/f:SI 1006 [739])
                        (const_int 4 [0x4])) [0  S4 A32])
                (mem:SI (plus:SI (reg/f:SI 1050 [738])
                        (const_int 4 [0x4])) [0  S4 A32]))
            (set (reg/f:SI 1006 [739])
                (plus:SI (reg/f:SI 1006 [739])
                    (const_int 8 [0x8])))
            (set (reg/f:SI 1050 [738])
                (plus:SI (reg/f:SI 1050 [738])
                    (const_int 8 [0x8])))
            (clobber (reg:SI 957))
            (clobber (reg:SI 958))
        ]) targets/nrf5x/bluetooth_utils.c:178 775 {movmem8b}
     (expr_list:REG_UNUSED (reg/f:SI 1050 [738])
        (expr_list:REG_UNUSED (reg/f:SI 1006 [739])
            (expr_list:REG_UNUSED (reg:SI 958)
                (expr_list:REG_UNUSED (reg:SI 957)
                    (nil))))))
targets/nrf5x/bluetooth.c:1246: confused by earlier errors, bailing out
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [espruino_1v91.18_microbit.elf] Error 1

```

gfwilliams, 2017-01-24:
I'm still using GCC 5 - it could be a bug in GCC6? I use this one and it works great: https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update

Travis uses a much older compiler which - yes - makes the files too big.

Personally I don't use Vagrant at all, so I'm not sure if that could be doing something that's causing problems? Many people do seem to have built this for nRF51 without issue though

OwenBrotherwood, 2017-01-24:
I will use the version you use and report back

OwenBrotherwood, 2017-01-25:
Today (just now) it worked after a pull from the repro with vagrant and version 6 of the gcc

closed 2017-01-25 by OwenBrotherwood

OwenBrotherwood, 2017-01-25:
Still investigating but the problem seems to be with doing 2 makes, one after the other.

I first compiled for NRF52 and it worked, also IRL on the board.
Then I tried to compile for MICROBIT which failed and therefor this issue.

Today, I started with a new clone and make MICROBIT first: but this time when I tried to compile NRF52, that failed.

A make clean between targets solves the problem.

(There are probably not so many people compiling MICROBIT and NRF52 and therefor the issue has not risen before or that people are using make clean between builds.)

gfwilliams, 2017-01-25:
Ahh, thanks for the update. I'll make a note - `make clean` is a pretty standard thing to try when you're having build errors though.

OwenBrotherwood, 2017-01-25:
Yep: I should really have thought about it but one forgets things sometimes.
The Vagrant file is great for quick setup of a development environment: I will be using it for Espruino as we are also using for other systems as it is the best method to remove unknowns as os / requirements and of course the gcc