espruino/Espruino #1035: New build options for ESP8266

MaBecker, 2017-01-26:
Hi @gfwilliams,

like to set ESP releases with this make options by default: 

    RELEASE=1  ESP8266_BOARD=1 USE_NET=1 USE_TELNET=1 USE_GRAPHICS=1 USE_CRYPTO=1 

and ESP8266_BOARD.py with 1600 variables to get a nice mix of Vars and Heap:

    ={ "free": 1573, "usage": 27, "total": 1600, "history": 0 }

    ={
      "sdkVersion": "2.0.0(5a875ba)",
      "cpuFrequency": 160, "freeHeap": 10912, "maxCon": 10,
      "flashMap": "4MB:512/512",
      "flashKB": 4096,
      "flashChip": "0xe0 0x4016"
     }

This will work for ESP01 up to ESP12, as far as i understand.

Please let me know if you agree, so  I can create the PRs for this.

gfwilliams, 2017-01-27:
That sounds good to me. Do you need crypto though, given it won't do TLS with just that?

Just wondering as I know a few people were using the filesystem stuff - that might be a better one to use instead?

MaBecker, 2017-01-27:
* module WS is bundled with crypto [var crypto = require('crypto');](https://github.com/espruino/EspruinoDocs/blob/master/modules/ws.js#L65)

* filesystem is a big one, have to check

gfwilliams, 2017-01-27:
Ahh, good point - yes, Websockets would get a lot more use than filesystem :)

MaBecker, 2017-01-31:
just created PR `#1044 neat build settings for ESP8266 boards`

MaBecker, 2017-01-31:
No This is too much - closed #1044, only want this change: 

https://github.com/MaBecker/Espruino/commit/3a2eb5f85aac66a8aa721f361d960c685dd54e61

@gfwilliams - please add label ESP8266

gfwilliams, 2017-01-31:
sure - thanks

gfwilliams, 2017-03-09:
Just committed this. Release is not there by default - you must use it on the command-line for a release build like you do with the other boards though

closed 2017-03-09 by gfwilliams