espruino/Espruino #1037: load() causes endless loops if "No code in flash!"

MaBecker, 2017-01-26:
@gfwilliams - please set label as ESP8266

    >load();
    =undefined
    No code in flash!
    ERROR: Not connected to the internet
    ERROR: Not connected to the internet
    ......

gfwilliams, 2017-01-26:
Wow, odd. Does it come back after a power cycle? `load()` shouldn't write anything to flash so it shouldn't be able to do anything permanent

MaBecker, 2017-01-26:
A hardware reset  or power off/on brings it back

MaBecker, 2017-01-26:
I have no idea where to start digging to fix this, do you have any hints ?

gfwilliams, 2017-01-27:
Not really - but maybe add some prints to `jswrap_flash.c` to see if you can tell where it ends up looping?

MaBecker, 2017-01-28:
Hi @tve,

is ist possible to force a stack dump in function [networkGetFromVarIfOnline](https://github.com/espruino/Espruino/blob/master/libs/network/network.c#L253) to analyse the call stack ?

I tried a div by 0 : causes a reset without stack dump

```
>load();
=undefined
No code in flash!
ERROR: Not connected to the internet
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

wilberforce, 2017-02-01:
https://github.com/espruino/Espruino/blob/9ec75f55ce4391550f8895128be9f22d77c6f9c1/src/jswrap_flash.c#L502

gfwilliams, 2017-02-01:
But does that explain why it keeps happening over and over?

Surely if it crashed it'd just reboot as if power was re-applied?

Perhaps it's an uninitialised variable somewhere. If the ESP8266 does a 'warm' reboot then I guess RAM contents will no longer be 0, and that might throw something off.

It still doesn't really explain why 'save()' would work, and a 'load()' doesn't. They should really be identical, since 'save()' writes to flash but doesn't change the RAM contents, and `load()` isn't changing RAM contents at this point either

MaBecker, 2017-02-01:
This version still works:

```
>load()
=undefined
No code in flash!
process.env.VERSION
="1v90.2079"
process.env.GIT_COMMIT
="f39dd3b75668f059487d6cbe1ba8fa6463885668"
```

commit f39dd3b75668f059487d6cbe1ba8fa6463885668
Author: Gordon Williams <gw@pur3.co.uk>
Date:   Mon Dec 19 18:52:27 2016 +0000

There is only one commit for /src/jsinteractive.c after that, which is 

https://github.com/espruino/Espruino/commit/db368a9dba7b999394f004913a4657e859daa35e#diff-a84215c1a9e81a0bd82d774d9354b71b

    git clone and git --hard reset db368a9 

and you have a version where load() is broken....

one step back - and load() is working

    git clone and git --hard reset  b18acf3

So something in this added function breaks load() !

       jsvSoftKill();
       jsvKill();                <-- this 
       jshReset();
       jsvInit();               <-- or this 
       jsfLoadStateFromFlash();
       jsvSoftInit();

gfwilliams, 2017-02-01:
Thanks for narrowing that down.

Most likely it's something in ESP8266 that allocates a JsVar but then doesn't free it on the `_kill` handler then.

jsvInit and jsvKill and basically totally clearing the contents of memory, so if some code were still trying to use something in that memory it'd break.

To be honest it would have broken anyway after a `load()` that completed successfully I guess?

MaBecker, 2017-02-01:
did one more test: it is jsvInit();

gfwilliams, 2017-02-01:
Can you try compiling again with a much lower variable count in the BOARD.py file?

Is it possible that you've increased the variable count so much that the variables are now overlapping with the execution stack?

MaBecker, 2017-02-01:
sure , set vars to 1000, still loops

gfwilliams, 2017-02-01:
Odd - please could you add some prints to jsvInit and try and see where it dies?

jumjum123, 2017-02-01:
I had a similiar problem some time ago. Sorry but I don't remember all details.
At the end it turned out, that magic number was written to flash, but no code (or code was overwritten, corrupted, ..)
After reading magic number, Espruino tried to read code and failed, caused a restart, ran into same problem again, .... 
I've in mind code is stored with pointer and length, and those are not checked (????)

MaBecker, 2017-02-01:
code section:

```
    if ((s&JSIS_TODO_FLASH_LOAD) == JSIS_TODO_FLASH_LOAD) {
      jsiStatus &= (JsiStatus)~JSIS_TODO_FLASH_LOAD;
      jsiSoftKill();
      jspSoftKill();
      jsvSoftKill();
jsvKill();
      jshReset();
jsvInit();
      jsfLoadStateFromFlash();
      jsvSoftInit();
      jspSoftInit();
      jsiSoftInit(false /* not been reset */);
    }
```

add function like `jsiConsolePrintf("> name()\n");` for IN and   `jsiConsolePrintf("< name()\n");` for OUT
```
> jsvInit()           in
> jsvSoftInit()       in
< jsvSoftInit()       out
< jsvInit()           out
No code in flash!
> jsvSoftInit()     in
< jsvSoftInit()     out
> jspSoftInit()     in 
< jspSoftInit()     out
ERROR: Not connected to the internet   lost.....
```

looks like it dies when calling ` jsiSoftInit(false /* not been reset */);`  ?!?

wilberforce, 2017-02-01:
dies?

I thought i was as endless loop?

Can you output the looping - there should be in and out calls within the loop if the correct part has the debugs?

MaBecker, 2017-02-01:
dies : loops : have to reboot to use it again.....

only added `jsiConsolePrintf()` from top to `loop`

sure could do the same from bottom to loop

tve, 2017-02-02:
> I had a similiar problem some time ago. Sorry but I don't remember all details.
> At the end it turned out, that magic number was written to flash, but no code (or code was overwritten, corrupted, ..) [...]

I'm a bit confused by the context of this "bug". I'm pretty sure that there is nothing esp8266 specific in load(). I believe the sanity check on what gets loaded are pretty thin, so there like jumjum I would expect garbage-in garbage-out, which may well result in a crash. The reboot will do the same thing as load(), so you get an infinite loop, no?

The init sequence is esp8266 specific, so it's possible that there is no bug in load() but that it triggers a re-init and that may break on the esp8266. I know that a number of thing simply are different due to the way the Espressif SDK works as well as due to the fact that there's wifi with a console right off-the-bat.

I think it would be really helpful to understand whether there is an issue with save()/load() when used correctly vs. when there is uninitialized stuff left in flash.

I'm using setBootCode() myself, which is different from save()/load(), so this is not an area I have looked into much.

gfwilliams, 2017-02-02:
@MaBecker ahh - I didn't realise it wasn't actually crashing in `jsvInit`.

To me, that points specifically to some ESP8266 initialistation - specifically something that gets initialised and uses JsVars, but then doesn't de-initialise on `kill`.

### What happened before

`load()` failed, left everything as it was before

### What happens now

`load()` failed, but everything gets cleared to 'reset' state. Any variables that were left allocated by ESP8266 code are now set to 'undefined', which will almost certainly cause a crash when they are next used.

To be honest I'm amazed that it doesn't crash in other cases as well, since `reset()` should do a very similar thing to a `load()` that fails.

MaBecker, 2017-02-02:
ok, one more test.

1. starting with empty flash for code
```
>reset();
=undefin> jsvInit()
> jsvSoftInit()
< jsvSoftInit()
< jsvInit()
> jspSoftInit()
< jspSoftInit()
> jsiSoftInit()
< jsiSoftInit()
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v90.2084 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
> 
```

2. dump() and save() 
```
>dump();
pinMode(D0, "input_pullup", true);
pinMode(D4, "input_pullup", true);
pinMode(D5, "input_pullup", true);
pinMode(D12, "input_pullup", true);
pinMode(D13, "input_pullup", true);
pinMode(D14, "input_pullup", true);
pinMode(D15, "input_pullup", true);
=undefined
>save();
=undefined
> jsvClearEmptyVarList()
< jsvClearEmptyVarList()
Erasing Flash.....
Writing....
Compressed 25600 bytes to 1066
Checking...
D> jsvSoftInit()
< jsvSoftInit()
> jspSoftInit()
< jspSoftInit()
> jsiSoftInit()
< jsiSoftInit()
> 
```

3. manual load()
```
>load();
=undefined
> jsvClea> jsvInit()
> jsvSoftInit()
< jsvSoftInit()
< jsvInit()
Loading 1066 bytes from flash...
> jsvSoftInit()
< jsvSoftInit()
> jspSoftInit()
< jspSoftInit()
> jsiSoftInit()
< jsiSoftInit()
> 
```

4. hardware reset
```
> jsvInit()
> jsvSoftInit()
< jsvSoftInit()
< jsvInit()
> jspSoftInit()
< jspSoftInit()
> jsvClearEmptyVarList()       <- this is diffent to a manual load()
< jsvClearEmptyVarList()
Loading 1066 bytes from flash...
> jsvSoftInit()
< jsvSoftInit()
> jspSoftInit()
< jspSoftInit()
> jsiSoftInit()
< jsiSoftInit()
>
```

The trace of used function differs between manual load and a load after reset.

gfwilliams, 2017-02-02:
Yes - that's what I pointed to above.

If you think about it, if a variable was used by WiFi, and you save everything, then load it again, there's a high chance that exact same variable will be loaded back and will be fine.

however if it's used by WiFi and you clear everything, it'll break.

The fact that `load()` fails now is a symptom rather than the underlying cause. Sure, you could just remove that code, but the code would still be horribly broken and would likely fail quite soon with some other problem

closed 2017-07-21 by MaBecker