espruino/Espruino #1040: micro:bit may have a problem with first connection

OwenBrotherwood, 2017-01-28:
The micro:bit(s) I am testing seem to have the same form of problem with first connection.
It seems that when connecting over usb (windows 10), that the board does not react with a normal Espruino logo (etc).
It first after one presses reset that the logo comes and things get to normal.
Tested with released firmware and github make.
EspruinoWeb and with espurino --list and -p 

With bluetooth, no way to continue at the moment, but I feel that if usb connection is solved, the bluetooth will slacken up.

```

C:\Users\owen\source\repos\github\espruino\EspruinoWebIDE>espruino --list
Espruino Command-line Tool 0.0.30
-----------------------------------

PORTS:
  d7:af:86:be:70:29 (Espruino MICROBIT)
  COM3 (Microsoft)
  COM4 (mbed)

C:\Users\owen\source\repos\github\espruino\EspruinoWebIDE>espruino -p COM4
Espruino Command-line Tool 0.0.30
-----------------------------------

Connecting to 'COM4'
No result found - just got ""
Connected
``` 
**Press reset**
```

_____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v91 Copyright 2016 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

```

OwenBrotherwood, 2017-01-28:
> Request board details on connect (on)
The request seems to be the problem, with off, it is possible to connect via usb and bluetooth without problems

OwenBrotherwood, 2017-01-28:
I think the following console log may be of use, Request board details on connect (on)

```
CONSOLE

The last 100 lines of log messages made by the Web IDE. This is only useful when trying to debug potential problems with the IDE or Espruino board itself.

BT> Sending "\r"
BT> Sending "\r"
BT> Sending "\r"
SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":true,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"setWatch(function(e) {\n  if (e.state) console.log(\"Released\");\n  else console.log(\"Pressed\");\n}, BTN1, {repeat:true, debounce:20, edge:\"both\"});\n\n","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":false,"ENV_ON_CONNECT":false,"FONT_SIZE":12,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":true,"MINIFICATION_Unreachable":true,"MINIFICATION_Unused":true,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"ESPRIMA","MODULE_URL":"http://www.espruino.com/modules","NPM_MODULES":false,"RESET_BEFORE_SEND":false,"SAVE_ON_SEND":0,"SERIAL_AUDIO":"0","SERIAL_TCPIP":"","SERIAL_THROTTLE_SEND":false,"SET_TIME_ON_WRITE":false,"SHOW_CODE_LINK_ICON":false,"SHOW_WEBCAM_ICON":0,"STORE_LINE_NUMBERS":true,"Sound_Error":"","Sound_Info":"","Sound_Success":"","Sound_Warning":"","Speak_Error":false,"Speak_Info":false,"Speak_Success":false,"Speak_Warning":false,"UI_MODE":"Normal"}
Config.ENV_ON_CONNECT => false
SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":true,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"setWatch(function(e) {\n  if (e.state) console.log(\"Released\");\n  else console.log(\"Pressed\");\n}, BTN1, {repeat:true, debounce:20, edge:\"both\"});\n\n","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":false,"ENV_ON_CONNECT":false,"FONT_SIZE":12,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":true,"MINIFICATION_Unreachable":true,"MINIFICATION_Unused":true,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"ESPRIMA","MODULE_URL":"http://www.espruino.com/modules","NPM_MODULES":false,"RESET_BEFORE_SEND":false,"SAVE_ON_SEND":0,"SERIAL_AUDIO":"0","SERIAL_TCPIP":"","SERIAL_THROTTLE_SEND":true,"SET_TIME_ON_WRITE":false,"SHOW_CODE_LINK_ICON":false,"SHOW_WEBCAM_ICON":0,"STORE_LINE_NUMBERS":true,"Sound_Error":"","Sound_Info":"","Sound_Success":"","Sound_Warning":"","Speak_Error":false,"Speak_Info":false,"Speak_Success":false,"Speak_Warning":false,"UI_MODE":"Normal"}
Config.SERIAL_THROTTLE_SEND => true
Disconnect callback...
WARNING: [notify_warn] Disconnected
>>> Disconnected
>>> Connecting...
ForceThrottle option is set - set Slow Write = true
ERROR: Unable to open device (connectionInfo=undefined)
ERROR: [notify_error] Connection Failed.
>>> Connection Failed.
ERROR: [notify_error] Connection Failed.
>>> Connection Failed.
>>> Connecting...
ForceThrottle option is set - set Slow Write = true
ERROR: Unable to open device (connectionInfo=undefined)
ERROR: [notify_error] Connection Failed.
>>> Connection Failed.
ERROR: [notify_error] Connection Failed.
>>> Connection Failed.
>>> Connecting...
ForceThrottle option is set - set Slow Write = true
Connected [object Object]
Device found (connectionId=undefined)
[success] Connected to port d7:af:86:be:70:29
>>> Connected to port d7:af:86:be:70:29
BT> Sending "\r"
BT> Sending "\r"
SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":true,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"setWatch(function(e) {\n  if (e.state) console.log(\"Released\");\n  else console.log(\"Pressed\");\n}, BTN1, {repeat:true, debounce:20, edge:\"both\"});\n\n","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":false,"ENV_ON_CONNECT":false,"FONT_SIZE":12,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":true,"MINIFICATION_Unreachable":true,"MINIFICATION_Unused":true,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"ESPRIMA","MODULE_URL":"http://www.espruino.com/modules","NPM_MODULES":false,"RESET_BEFORE_SEND":false,"SAVE_ON_SEND":0,"SERIAL_AUDIO":"0","SERIAL_TCPIP":"","SERIAL_THROTTLE_SEND":false,"SET_TIME_ON_WRITE":false,"SHOW_CODE_LINK_ICON":false,"SHOW_WEBCAM_ICON":0,"STORE_LINE_NUMBERS":true,"Sound_Error":"","Sound_Info":"","Sound_Success":"","Sound_Warning":"","Speak_Error":false,"Speak_Info":false,"Speak_Success":false,"Speak_Warning":false,"UI_MODE":"Normal"}
Config.SERIAL_THROTTLE_SEND => false
Disconnect callback...
WARNING: [notify_warn] Disconnected
>>> Disconnected
>>> Connecting...
Set Slow Write = true
Connected [object Object]
Device found (connectionId=undefined)
[success] Connected to port d7:af:86:be:70:29
>>> Connected to port d7:af:86:be:70:29
BT> Sending "\r"
BT> Sending "\r"
Disconnect callback...
WARNING: [notify_warn] Disconnected
>>> Disconnected
>>> Connecting...
Set Slow Write = true
[object Object]
Connected [object Object]
Device found (connectionId=1)
[success] Connected to port COM4
>>> Connected to port COM4
Disconnect callback...
WARNING: [notify_warn] Disconnected
[... cut ...]
Set Slow Write = true
Connected [object Object]
Found a prompt... great!
>>> Sending...
---> "\u0010console.log(\"<\",\"<<\",JSON.stringify(process.env),\">>\",\">\")\n"
BT> Sending "\u0010console.log(\"<"
BT> Sending "\",\"<<\",JSON.str"
BT> Sending "ingify(process."
BT> Sending "env),\">>\",\">\")\n"
>>> Sent
WARNING: No result found - just got ""
Device found (connectionId=undefined)
[success] Connected to port d7:af:86:be:70:29
>>> Connected to port d7:af:86:be:70:29
BT> Sending "\r"
BT> Sending "\r"

```

OwenBrotherwood, 2017-01-28:
It is probably a good idea to use as few software components as possible, so espruino and a json to document.

"ENV_ON_CONNECT": true or false : same result, no response from micro:bit when using COM4
This is different from the EspruinoWeb (electron), where false gave a console to work on.

I continue to look thru what can be done to document or even fix something.

**espruino  -j config.json**
```
C:\Users\owen\source\repos\github\espruino>espruino  -j config.json
0.0.30
Espruino Command-line Tool 0.0.30
-----------------------------------

Acorn library not found - you'll need it for compiled code
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/libs/targz.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/libs/utf8.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/espruino.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/codeWriter.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/config.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/env.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/flasher.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/modules.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/notifications.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial_audio.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial_chrome.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial_noble.js
'noble' module couldn't be loaded, no node.js Bluetooth Low Energy
 Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (C:\Users\owen\AppData\Roaming\npm\node_modules\espruino\node_modules\bluetooth-hci-socket\lib\usb.js:70:11)
    at BluetoothHciSocket.bindRaw (C:\Users\owen\AppData\Roaming\npm\node_modules\espruino\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (C:\Users\owen\AppData\Roaming\npm\node_modules\espruino\node_modules\noble\lib\hci-socket\hci.js:99:35)
    at NobleBindings.init (C:\Users\owen\AppData\Roaming\npm\node_modules\espruino\node_modules\noble\lib\hci-socket\bindings.js:83:13)
    at new Noble (C:\Users\owen\AppData\Roaming\npm\node_modules\espruino\node_modules\noble\lib\noble.js:50:18)
    at Object.<anonymous> (C:\Users\owen\AppData\Roaming\npm\node_modules\espruino\node_modules\noble\index.js:4:18)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial_nodeserial.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial_socket.js
No chrome.sockets - serial_socket disabled
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial_websocket.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial_web_bluetooth.js
No navigator.bluetooth - Web Bluetooth not enabled
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/serial_winnus.js
Disable Web Bluetooth as we have Winnus instead
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/settingsAbout.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/status.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/terminal.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/core/utils.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/assembler.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/boardJSON.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/compiler.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/getGitHub.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/localModules.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/minify.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/npmModules.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/saveOnSend.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/setTime.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/uiMode.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/unicode.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/versionChecker.js
Found C:\Users\owen\AppData\Roaming\npm\node_modules\espruino/plugins/_examplePlugin.js
Initialising CodeWriter
Initialising Config
Initialising Env
Initialising Flasher
Initialising Modules
[... 58 lines cut ...]
    "ENV_ON_CONNECT": true,
    "MINIFICATION_LEVEL": "",
    "MINIFICATION_Mangle": true,
    "MINIFICATION_Unreachable": true,
    "MINIFICATION_Unused": true,
    "MINIFICATION_Literal": true,
    "MINIFICATION_DeadCode": true,
    "MODULE_AS_FUNCTION": false,
    "MODULE_EXTENSIONS": ".min.js|.js",
    "MODULE_MINIFICATION_LEVEL": "ESPRIMA",
    "MODULE_URL": "http://www.espruino.com/modules",
    "NPM_MODULES": false,
    "RESET_BEFORE_SEND": true,
    "SAVE_ON_SEND": 0,
    "SERIAL_AUDIO": 0,
    "SERIAL_TCPIP": "",
    "SERIAL_THROTTLE_SEND": false,
    "SET_TIME_ON_WRITE": false,
    "STORE_LINE_NUMBERS": true,
    "UI_MODE": "Normal",
    "WEB_BLUETOOTH": true
  }
}

```

OwenBrotherwood, 2017-01-28:
I apparently missed the tick in the GUI: the cmd tool and web have same result with the following, access to the Espruino cmd prompt.
This gives a good base for looking for a possible fix

```
"ENV_ON_CONNECT": true,
"SERIAL_THROTTLE_SEND" : true

```

OwenBrotherwood, 2017-01-29:
I double checked on nRF52-DK and no problems with "ENV_ON_CONNECT": true, "SERIAL_THROTTLE_SEND" : false
I am tempted to look thru the code and see if there is a place where connected with no prompt can have a code addition for a retry with some form of additional logic to see if env can be obtained with a prompt.
As such, this would be a PR from the community so you can concentrate on Espruino and not other BOARD's

OwenBrotherwood, 2017-01-29:
Having spent a bit on the core js files of EspruinoTools, I go over to the actual firmware that is on the micro:bit with a flash of code from lattest github.
It should be noted that the usb light blinks with characters sent from esprunino but no reply is received.

---

I enabled console.log for what is actually sent, should probably be on when in verbose
```

    // actually write data
    console.log("Sending block "+JSON.stringify(d)+", wait "+split.delay+"ms");

```

OwenBrotherwood, 2017-01-29:
Side issue from current github code:
`
vagrant@vagrant-ubuntu-trusty-64:/vagrant$ make clean;MICROBIT=1 DEBUG=1 make
`
Gives:
```
LD espruino_1v91.31_microbit.elf
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: espruino_1v91.31_microbit.elf section `.text' will not fit in region `FLASH'
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 114300 bytes
collect2: error: ld returned 1 exit status
make: *** [espruino_1v91.31_microbit.elf] Error 1

```
Wish micro:bit had more flash/ram : continues without debug

OwenBrotherwood, 2017-01-29:
Status:
micro:bit can work with the following:
http://www.espruino.com/files/espruino_1v91.zip
http://www.espruino.com/files/espruino_ide_win64_0v65.9.exe
and
[image]
If one wishes bluetooth.
If one wishes to have 
[image]
It is possible with usb: but the micro:bit needs a reset before things start to work (process info not available)

---

At some stage, I look more into the problem as I gain experience in the code.

OwenBrotherwood, 2017-01-29:
Notepad:
- possibilities in having "stuff" on SD Flash: I read somewhere that there is something that may allow this
 - make clean;MICROBIT=1 RELEASE=1 VARIABLES=10 USE_FILESYSTEM=1 make
- allow for place for debug: no room :(
 - make clean;MICROBIT=1 DEBUG=1 VARIABLES=1 make

olliephillips, 2017-01-29:
I haven't read all of this issue, but first connection display of espruino 'ASCII' logo etc can be hit and miss on many boards. The acid test of a correct connection is do you have console access and can you send code. Since you can reset and see the Espruino ASCII then you have one or both?

OwenBrotherwood, 2017-01-30:
nRF52-DK (and soon puck.js) are used as reference and a correct setup for developement is established.
vagrant ubuntu and windows 10 for end user tests

gfwilliams, 2017-01-30:
You won't see the startup logo on a lot of boards - that doesn't have anything to do with the request for environment variables.

The logo is sent at boot time (and only if no code is saved), and on micro:bit the USB-Serial interface will take some time to establish, during which time it will be ignoring anything sent to it. Short of intentionally slowing Espruino's boot time down just so the logo can be made to appear, I don't think there's anything that can be done about it

gfwilliams, 2017-01-30:
As @olliephillips says, the real test is whether you can actually write characters on the left hand side of the IDE and make stuff happen

OwenBrotherwood, 2017-01-30:
yes: acid test is left hand side active.
if one does not have the following, there is no active left side active.

I return to the micro:bit problem after nRF52-DK and puck.js tests and some thought.

"ENV_ON_CONNECT": false,
"SERIAL_THROTTLE_SEND" : true

gfwilliams, 2017-02-03:
Can I close this? Not getting a logo on reset is to be expected on Micro:Bit because of what I said above

OwenBrotherwood, 2017-02-04:
yes close it.
after some more work on micro:bit, I may have a summary.
unfortunatly, my drains have gone so I have to dig up the garden this weekend.
I will improve on the using micro:bit with Espurino gif and ensure that slack micro:bit slack has a nice view of Espurino when I have ensured that there are no issues.
On tuesday, the group of volenteers that teach kids in their spare time are meeting and will be looking at the microbit
I will be showint the micro:bit with espurino and the puck.js

closed 2017-02-04 by OwenBrotherwood

OwenBrotherwood, 2017-02-04:
By the way, I have to repeat the acid test in controlled environment with documentation in the form of video with relation to the acid test.

> yes: acid test is left hand side active.
> if one does not have the following, there is no active left side active.
> 
> I return to the micro:bit problem after nRF52-DK and puck.js tests and some thought.
> 
> "ENV_ON_CONNECT": false,
> "SERIAL_THROTTLE_SEND" : true